norlab-ulaval/libpointmatcher #426: Better location for libpointmatcher installation

maximecharriere, 2020-12-09:
Hello, 
when configuring **libpointmatcher** with CMake, the variable `CMAKE_INSTALL_PREFIX` will be set by default to `C:/Program Files (x86)/libpointmatcher` (on Windows). It's in this directory that will be generated folders `include` and `lib`.
Sometimes the user doesn't have write permissions to the `C:/Program Files (x86)` directory. For example if he launches cmake from the cmd or PowerShell without administrator privileges. 
And it seems more logical to me to generate these two folders in the directory where the user wants to build his project (Usually the `build` folder).

I am not familiar with the creation of `CMakeLists.txt`, but can't we change something so the `include` and `lib` folders are generated inside the `build` folder ?

pomerlef, 2020-12-09:
Have you tried just running `make` instead of `make install`? I think it will do you you want. The `make install` is mostly when you need to integrate libpm in multiple projects and don't want to manually locate the files each time.

maximecharriere, 2020-12-10:
Okay I see what you mean. I use Visual Studio so I don't use `make install`, but in my case I don't have to build the `INSTALL` project inside VS.

As I understand it, the `make install` will simply move files already in the `build` directory to a better sorted and cleaner directory.

closed 2020-12-10 by maximecharriere