norlab-ulaval/libpointmatcher #431: Filters, Matcher, Minimizer, Checkers, Inspector, Logger documentation

maximecharriere, 2021-01-05:
Hello,
I'm beginning with libpointmatcher, and I want to find docs about
- readingDataPointsFilters
- referenceDataPointsFilters
- matcher
- outlierFilters
- errorMinimizer
- transformationCheckers
- inspector
- logger  

I found a list of components name to use in the YAML file [here](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Configuration.md). But this list is only partially complete, because if we look at all the Datapoint Filters documentation [here](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md) we can notice that there are 7 filters missing in the list:
> #### Down-sampling
> 5. [Distance Limit Filter](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#distancelimithead)    
> ---
> 11. [Voxel Grid Filter](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#voxelgridhead) (**deprecated**)    
> 12. [Octree Grid Filter](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#octreegridhead)
> 13. [Normal Space Sampling (NSS) Filter](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#nsshead)
> 14. [Covariance Sampling (CovS) Filter](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#covshead)
> 15. [Spectral Decomposition Filter (SpDF)](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#spdfhead)
> #### Descriptor Augmenting
> 6. [Saliency Filter](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/DataFilters.md#saliencyhead)

For other modules than DataPointsFilters, I only found the [website of Norlab](https://norlab.ulaval.ca/libpointmatcher-modules/), but it doesn't seem to be up to date.

---  

**Where can I find up-to-date documentation about all modules, their component name and associated parameters to be written to the YAML file ?**

pomerlef, 2021-01-05:
The documentation lags behind some implementations. Luckily, there is a documentation embedded in the library. You can run

```
$ ./examples/list_modules
```

to see what modules are compiled on your system.

maximecharriere, 2021-01-05:
Thank you! I will use the doc generated by list_modules
If anyone is interested, [here is the result](https://gist.github.com/maximecharriere/c4b7e3f901994f29cb361de98cf25b72)

closed 2021-01-05 by maximecharriere