norlab-ulaval/libpointmatcher #432: failed linking a simple libpointmatcher project under Ubuntu

josepforest, 2021-01-11:
Hello,

I'm trying to compile the demo_cmake project under ubuntu 20.04. All dependencies are satisfied and the library compiles and installs smoothly. However, when trying to compile this simple application, I get a lot of linker errors complaining unresolved references to different boost functions:

/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::is_regular_file(boost::filesystem::path const&)':
/usr/include/boost/filesystem/operations.hpp:473: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::initial_path()':
/usr/include/boost/filesystem/operations.hpp:643: undefined reference to `boost::filesystem::detail::initial_path(boost::system::error_code*)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::complete(boost::filesystem::path const&)':
/usr/include/boost/filesystem/operations.hpp:530: undefined reference to `boost::filesystem::absolute(boost::filesystem::path const&, boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::exists(boost::filesystem::path const&)':
/usr/include/boost/filesystem/operations.hpp:463: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::operator/(boost::filesystem::path&&, boost::filesystem::path const&)':
/usr/include/boost/filesystem/path.hpp:810: undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::exists(boost::filesystem::path const&)':
/usr/include/boost/filesystem/operations.hpp:463: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::operator/(boost::filesystem::path&&, boost::filesystem::path const&)':
/usr/include/boost/filesystem/path.hpp:810: undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `PointMatcherIO<double>::FileInfoVector::FileInfoVector(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/jforest/Baixades/programari/libpointmatcher/pointmatcher/IO.cpp:212: undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: /home/jforest/Baixades/programari/libpointmatcher/pointmatcher/IO.cpp:204: undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `PointMatcherIO<float>::FileInfoVector::FileInfoVector(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/jforest/Baixades/programari/libpointmatcher/pointmatcher/IO.cpp:212: undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: /home/jforest/Baixades/programari/libpointmatcher/pointmatcher/IO.cpp:204: undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(IO.cpp.o): in function `boost::filesystem::extension[abi:cxx11](boost::filesystem::path const&)':
/usr/include/boost/filesystem/convenience.hpp:36: undefined reference to `boost::filesystem::path::extension() const'
/usr/bin/ld: /usr/include/boost/filesystem/convenience.hpp:36: undefined reference to `boost::filesystem::path::extension() const'
/usr/bin/ld: /usr/include/boost/filesystem/convenience.hpp:36: undefined reference to `boost::filesystem::path::extension() const'
/usr/bin/ld: /usr/include/boost/filesystem/convenience.hpp:36: undefined reference to `boost::filesystem::path::extension() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/convert.dir/build.make:106: convert] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/convert.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

I browsed the internet for solutions but didn't actually find it. Looks like the library is not linked against libBoost, even when it is required in the CMakeLists.txt. I tried adding the include dirs and the target libraries for the boost in the CMakeLists.txt of the demo project with no luck. Honestly I'm a bit lost. Could you give me some directions on how can I overcome this problem ?

Thanks a lot in advance

Josep

pomerlef, 2021-01-11:
@AlexandreG87 can you have a look?

AlexandreG87, 2021-01-11:
@josepforest Hi! Here's a quick fix to your problem, you need to modify the `CMakeLists.txt` file in the `demo_cmake/` directory to look like this :

```cmake
cmake_minimum_required (VERSION 2.8)
project(convert)

add_compile_options(-std=c++11)

find_package(Boost REQUIRED COMPONENTS filesystem system)
find_package(libpointmatcher REQUIRED)

include_directories(${CMAKE_CURRENT_BINARY_DIR} ${libpointmatcher_INCLUDE_DIRS} ${Boost_INCLUDE_DIRS})

add_executable(${PROJECT_NAME} convert.cpp)

target_link_libraries(${PROJECT_NAME} ${libpointmatcher_LIBRARIES} ${Boost_LIBRARIES})
```

I tested it on Ubuntu 18.04 and 20.04 and it worked. Let me know if this doesn't solve your issue.

josepforest, 2021-01-12:
Quick and smooth, Alexandre !! Thanks a lot !! Issue solved.

El 11/1/21 a les 19:16, Alexandre Guénette ha escrit:

closed 2021-01-13 by AlexandreG87