norlab-ulaval/libpointmatcher #433: How to test on windows

821736960, 2021-01-12:
Hi, I have followed instructions to compile and install the libpointmatcher with VS 2019 on Win 10.  But I want to make sure that it can run successfully. So how to test?

pomerlef, 2021-01-12:
You can do this tutorial: https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ICPIntro.md

AlexandreG87, 2021-01-12:
Hi! If you want to make sure that everything is working as it should, you can also try to compile the unit tests by following this procedure using **cmake-gui**:

1. Set the variable `BUILD_TESTS` to `ON` or `TRUE`. If the variable is not present, just add it like you did with the other variables.
2. Rebuild the project after having configured and generated the cmake's files.
3. Run the following command : `utest/utest --path ../examples/data/` from a terminal in libpointmatcher's build directory.

I'd like to mention that I'm not a Windows user anymore and that I never used VS, so this procedure haven't been tested. It's similar to the one found in [Compiling and installing libpointmatcher on Ubuntu](https://libpointmatcher.readthedocs.io/en/latest/CompilationUbuntu/) tutorial, so it may not work.

821736960, 2021-01-13:
> Hi! If you want to make sure that everything is working as it should, you can also try to compile the unit tests by following this procedure using **cmake-gui**:
>
> 1. Set the variable `BUILD_TESTS` [...]

Thanks, my exe file locates at:  ~\build\utest\RelWithDebInfo\utest.exe,  and I run in Powershell :
 utest\RelWithDebInfo\utest --path ../examples/data/
It shows:
...
[----------] Global test environment tear-down
[==========] 72 tests from 13 test cases ran. (36224 ms total)
[  PASSED  ] 72 tests.
So that means I installed it correctlly?

821736960, 2021-01-13:
> You can do this tutorial: https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ICPIntro.md

https://github.com/CGAL/cgal/issues/5346#issue-784800927
Sorry, I meet some problems when linking libpointmatcher to CGAL, could you help? Thanks.

AlexandreG87, 2021-01-13:
> So that means I installed it correctly?

Not installed, but compiled successfully. Which means that the installation should be correct too.

821736960, 2021-01-13:
> > So that means I installed it correctly?
> 
> Not installed, but compiled successfully. Which means that the installation should be correct too.

Thanks. I meet some problems when linking libpointmatcher to CGAL, could you help?

AlexandreG87, 2021-01-13:
> Thanks. I meet some problems when linking libpointmatcher to CGAL, could you help?

Before going further with your CGAL request, were you able to do the tutorial? If so, that would mean that you've successfully compiled, installed and tested `libpointmatcher` on Windows, meaning that this issue could be closed. I invite you to open a new issue concerning your CGAL problem, so it would make it easier for others users that are encountering the same problem to find a solution. Thanks!

821736960, 2021-01-13:
> Before going further with your CGAL request, were you able to do the tutorial? If so, that would mean that you've successfully compiled, installed and tested `libpointmatcher` on Windows, meaning [...]

Sorry, I have not test the tutorial yet. I will try it as soon as possible!

821736960, 2021-01-14:
> Before going further with your CGAL request, were you able to do the tutorial? If so, that would mean that you've successfully compiled, installed and tested `libpointmatcher` on Windows, meaning [...]

Thanks for so detailed instructions! I have done the tutorial, it worked well.  I will open another issue.

closed 2021-01-14 by 821736960