norlab-ulaval/libpointmatcher #434: link to CGAL

821736960, 2021-01-14:
Sorry to bother.

Issue Details
I have installed CGAL 5.1.1 through vcpkg, openGR, libpointmatcher. And I want to compile the example: CGAL-5.1.2-examples-Point_set_processing_3.

For OpenGR wrapper, I just set OpenGR_DIR(at cmake-gui) as E:/OpenGR/build and the examples related to OpenGR can run well.

But for Libpointmatcher wrapper, I set (actually it set itself automatically) libpointmatcher_DIR(at cmake-gui) as E:/libpointmatcher-1.3.1/libpointmatcher-1.3.1/build , and the complile is failed.

Source Code
Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
Visual Leak Detector (VLD) is not found.
Using header-only CGAL
Targetting Visual Studio 16 2019
Target build enviroment supports auto-linking
Using VC toolset 142.
Generator uses intermediate configuration directory: $(Configuration)
Boost include dirs: E:/vcpkg/installed/x64-windows/include
Boost libraries:
USING DEBUG CXXFLAGS = '/DWIN32 /D_WINDOWS /GR /EHsc /Zi /Ob0 /Od /RTC1'
USING DEBUG EXEFLAGS = '/machine:x64 /debug /INCREMENTAL'
USING RELEASE CXXFLAGS = '/DWIN32 /D_WINDOWS /GR /EHsc /O2 /Ob2 /DNDEBUG'
USING RELEASE EXEFLAGS = '/machine:x64 /INCREMENTAL:NO'
NOTICE : the LAS reader test requires LASlib and will not be compiled.
CMake Error at E:/vcpkg/installed/x64-windows/share/cgal/CGAL_pointmatcher_support.cmake:3 (set_target_properties):
Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
"optimized". The INTERFACE_LINK_LIBRARIES property may contain
configuration-sensitive generator-expressions which may be used to specify
per-configuration rules.
Call Stack (most recent call first):
CMakeLists.txt:94 (include)

Configuring incomplete, errors occurred!
See also "E:/CGAL-5.1.2-examples/CGAL-5.1.2/examples/Point_set_processing_3/tt/CMakeFiles/CMakeOutput.log".
See also "E:/CGAL-5.1.2-examples/CGAL-5.1.2/examples/Point_set_processing_3/tt/CMakeFiles/CMakeError.log".

Environment
Operating system (Windows/64 bits):
Compiler: VS 2019, Cmake
CGAL version: 5.1.1

821736960, 2021-01-14:
#433  yesterday I compiled libpointmatcher and done the tutorial. Today the author of CGAL suggests me to upgrade my boost version from 1.66 to 1.71 to solve this issue. So I upgrade boost version to 1.71, and then compile libnabo, then libpointmatcher.

At this time I install boost 1.71with command:
`$ .\bootstrap.bat
 $ .\b2.exe install --prefix=build address-model=64`
Then install libnabo.

BUT as I begin to build  libpointmatcher:
Start CMake GUI, follow the same steps to configure the source and build folders for libpointmatcher, then click the Configure button.

It shows: Error in configuration process, project files may be invalid.
`Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
CMake Error at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR thread filesystem system
  program_options date_time)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindBoost.cmake:2162 (find_package_handle_standard_args)
  CMakeLists.txt:118 (find_package)

Configuring incomplete, errors occurred!
See also "E:/libpointmatcher-1.3.1/libpointmatcher-1.3.1/build/CMakeFiles/CMakeOutput.log".
See also "E:/libpointmatcher-1.3.1/libpointmatcher-1.3.1/build/CMakeFiles/CMakeError.log".`

maxGimeno, 2021-01-14:
You need to fill Boost_DIR with the path to <Boost_compiled_lib_dir>/cmake/Boost_1_73_0

821736960, 2021-01-14:
> You need to fill Boost_DIR with the path to <Boost_compiled_lib_dir>/cmake/Boost_1_73_0

E:\boost_1_73_0\boost_1_73_0\build\lib\cmake\Boost-1.73.0

The error still exists.

821736960, 2021-01-14:
> You need to fill Boost_DIR with the path to <Boost_compiled_lib_dir>/cmake/Boost_1_73_0

What is your cmake version？I go to see the cmake-3.17/Modules/FindBoost.cmake. It seems that the max boost version that cmake 3.17 can search is 1.72.0

821736960, 2021-01-15:
For the issue about couldnot find boost, I reinstalled boost 1.71 with exe files **rather than** .zip file with booststrap command. Now libpointmatcher can work again.

But the issue about compiling in CGAL still exists.

aguenette, 2021-01-17:
> Now libpointmatcher can work again.

This is great! :smile:

> But the issue about compiling in CGAL still exists.

I tried to compile libpointmatcher on a Windows VM, but without success. I'm not familiar enough with C++/CMake development on Windows. On the other hand, I tried to compile the example `Point_set_processing_3` on Ubuntu 18.04 and it only succeeded if libpointmatcher and libnabo were compiled as shared libraries. If you want to try this, you'll have to manually change the line 358 from `add_library(pointmatcher ${POINTMATCHER_SRC} ${POINTMATCHER_HEADERS} )` to `add_library(pointmatcher SHARED ${POINTMATCHER_SRC} ${POINTMATCHER_HEADERS} )` in the `CMakeLists file. txt` file, because it seems that this is the only way to do it since the recent changes made to the file. Regarding libnabo, just enable the `SHARED_LIBS` variable with cmake-gui. I really hope that this will work for you. If not, something might be broken with the new `CMakeLists.txt` file.