norlab-ulaval/libpointmatcher #438: MaxDensityDataPointsFilter not working

dagata-mining, 2021-01-27:
I'm binding libpointmatcher to CloudCompare I'm trying to implement all the different filters.
However when implementing MaxDensityDataPointsFilter I never get any subsampling out. I'm using the SurfaceNormalDataPointsFilter in first place then the MaxDensityDataPointsFilter 
such as:
`m_normalParams = PM::get().DataPointsFilterRegistrar.create(
		"SurfaceNormalDataPointsFilter",
		{
			{"knn", "5"},
			{"keepNormals", "1"},
			{"keepDensities", "1"},
			{"keepEigenValues", "0"},
			{"keepEigenVectors", "0"},
			{"keepMatchedIds", "0"},
		}
	);
filterParams = PM::get().DataPointsFilterRegistrar.create(
			"MaxDensityDataPointsFilter",
			{
				{"maxDensity", "10"},
			}
		);

`
Any Advice?

pomerlef, 2021-01-27:
My best advice would be to test your filter using our example code `conver.cpp`.  You can find it in `example/demo_cmake` and compile it from there.

It will take a yaml file and a point cloud as input. You can use the yaml file to test you configuration on a stand alone executable. In general, we use often `MaxDensityDataPointsFilter`, so it should be working. Tuning the parameter will depend on your point clouds.

dagata-mining, 2021-01-27:
Alright, 
Must be on my side then, just wanted to make sure there's no known issues with the densities descriptors.. 
Thanks!

closed 2021-01-27 by dagata-mining