norlab-ulaval/libpointmatcher #440: Boost errors when building simple ICP test program

JeffR1992, 2021-01-31:
I followed the instructions to build and install libpointmatcher for my Ubuntu 18.04 system [here](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationUbuntu.md), which worked without any issues (and passed all unit tests). However,  when trying to build a simple test program, I am getting a lot of build errors related to Boost. Here is the program I'm trying to build:
```
#include <iostream>
#include <pointmatcher/PointMatcher.h>

int main() {
	PointMatcher<float>::ICP icp;
    icp.setDefault();

	PointMatcher<float>::DataPoints data_points;

	std::cout << "ICP created.\n";

	return 0;
}
```

and my CMakeLists.txt is as follows (which looks similar the the tutorial given [here](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/LinkingProjects.md)):
```
cmake_minimum_required(VERSION 3.18.0)
project(test_project)

find_package(libpointmatcher REQUIRED)
include_directories(${libpointmatcher_INCLUDE_DIRS})
add_executable(icp_test icp_test.cpp)
target_link_libraries(icp_test ${libpointmatcher_LIBRARIES})

set(CMAKE_CXX_COMPILER g++)
```

Running `cmake ../src` I get the following:
```
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using libpointmatcher version 1.3.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alex/Desktop/icp_testing/build
```
And running `make` gives me the following:
```
Scanning dependencies of target icp_test
[ 50%] Building CXX object CMakeFiles/icp_test.dir/icp_test.cpp.o
[100%] Linking CXX executable icp_test
/usr/local/lib/libpointmatcher.a(DataPoints.cpp.o): In function `__static_initialization_and_destruction_0':
/usr/include/boost/system/error_code.hpp:206: undefined reference to `boost::system::generic_category()'
/usr/include/boost/system/error_code.hpp:208: undefined reference to `boost::system::generic_category()'
/usr/include/boost/system/error_code.hpp:210: undefined reference to `boost::system::system_category()'
/usr/local/lib/libpointmatcher.a(ICP.cpp.o): In function `PointMatcher<float>::ICPSequence::setMap(PointMatcher<float>::DataPoints const&)':
/usr/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::generic_category()'
/usr/local/lib/libpointmatcher.a(ICP.cpp.o): In function `PointMatcher<double>::ICPSequence::setMap(PointMatcher<double>::DataPoints const&)':
/usr/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::generic_category()'
/usr/local/lib/libpointmatcher.a(ICP.cpp.o): In function `PointMatcher<float>::ICPChainBase::loadFromYaml(std::istream&)':
/usr/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::generic_category()'
/usr/local/lib/libpointmatcher.a(ICP.cpp.o): In function `PointMatcher<double>::ICPChainBase::loadFromYaml(std::istream&)':
/usr/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::generic_category()'
/usr/local/lib/libpointmatcher.a(ICP.cpp.o): In function `PointMatcher<float>::ICP::computeWithTransformedReference(PointMatcher<float>::DataPoints const&, PointMatcher<float>::DataPoints const&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&)':
/usr/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::generic_category()'
/usr/local/lib/libpointmatcher.a(ICP.cpp.o):/usr/include/boost/thread/exceptions.hpp:51: more undefined references to `boost::system::generic_category()' follow
/usr/local/lib/libpointmatcher.a(ICP.cpp.o): In function `__static_initialization_and_destruction_0':
[... 341 lines cut ...]
/usr/local/lib/libpointmatcher.a(Exceptions.cpp.o): In function `__static_initialization_and_destruction_0':
/usr/include/boost/system/error_code.hpp:206: undefined reference to `boost::system::generic_category()'
/usr/include/boost/system/error_code.hpp:208: undefined reference to `boost::system::generic_category()'
/usr/include/boost/system/error_code.hpp:210: undefined reference to `boost::system::system_category()'
CMakeFiles/icp_test.dir/icp_test.cpp.o: In function `__static_initialization_and_destruction_0(int, int)':
icp_test.cpp:(.text+0x114): undefined reference to `boost::system::generic_category()'
icp_test.cpp:(.text+0x120): undefined reference to `boost::system::generic_category()'
icp_test.cpp:(.text+0x12c): undefined reference to `boost::system::system_category()'
CMakeFiles/icp_test.dir/icp_test.cpp.o: In function `boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const':
icp_test.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0xbb): undefined reference to `boost::system::generic_category()'
icp_test.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0xf6): undefined reference to `boost::system::generic_category()'
CMakeFiles/icp_test.dir/icp_test.cpp.o: In function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
icp_test.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xbc): undefined reference to `boost::system::generic_category()'
icp_test.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xf7): undefined reference to `boost::system::generic_category()'
icp_test.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x1de): undefined reference to `boost::system::generic_category()'
collect2: error: ld returned 1 exit status
CMakeFiles/icp_test.dir/build.make:105: recipe for target 'icp_test' failed
make[2]: *** [icp_test] Error 1
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/icp_test.dir/all' failed
make[1]: *** [CMakeFiles/icp_test.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2
```

I've also tried including and linking against the Boost `system` and `filesystem` libraries in my CMakeLists.txt, but this hasn't fixed the errors either. Any assistance would be much appreciated, thanks.

aguenette, 2021-01-31:
Hi @JeffR1992! I'm using Ubuntu 18.04 too and I used this CMakeLists.txt file and it worked for me. Does your file looked like this after trying to include and link against Boost?

```cmake
cmake_minimum_required(VERSION 3.18.0)
project(test_project)

find_package(libpointmatcher REQUIRED)
find_package(Boost REQUIRED COMPONENTS filesystem)

include_directories(${libpointmatcher_INCLUDE_DIRS} ${Boost_INCLUDE_DIR})
add_executable(icp_test icp_test.cpp)
target_link_libraries(icp_test ${libpointmatcher_LIBRARIES} ${Boost_LIBRARIES})

set(CMAKE_CXX_COMPILER g++)
```

JeffR1992, 2021-01-31:
Thanks @aguenette, I had thought that I included and linked `Boost` just as you had done above. But surprisingly using your `CMakeLists.txt` produces no compile errors, so I must have messed something up somewhere when linking `Boost` myself!

As an additional question, since the tutorial for linking `libpointmatcher` doesn't show any requirements for also linking `Boost`, does this mean that the tutorial is incorrect?

aguenette, 2021-01-31:
> Thanks @aguenette, I had thought that I included and linked Boost just as you had done above. But surprisingly using your CMakeLists.txt produces no compile errors, so I must have messed something up somewhere when linking Boost myself!

@JeffR1992 You're welcomed! 🙂

> As an additional question, since the tutorial for linking `libpointmatcher` doesn't show any requirements for also linking Boost, does this mean that the tutorial is incorrect?

You're right, but it hasn't been an issue before. PR #437 is supposed to solve that. But, the example should be updated anyway.

pomerlef, 2021-02-01:
PR #437 as been merged. @JeffR1992  could you test on your side and If you can send a quick PR to fix the doc with what worked for you?

We have limited resources for Windows on our side.

aguenette, 2021-02-07:
@pomerlef I've just tested the new fix and it worked without having to specify `find_package(Boost REQUIRED COMPONENTS filesystem)`. So, it should work for @JeffR1992 as well since he' also on Ubuntu 18.04.

pomerlef, 2021-02-08:
So I 'll close this issue. Reopen it if needed!

closed 2021-02-08 by pomerlef