norlab-ulaval/libpointmatcher #442: Libnabo Cmake problem

avgilias, 2021-02-06:
Hello, 
I am trying to generate libnabo, following step by step the tutorial and having the following probelm.
Cmake Error : 
CMake Error at CMakeLists.txt:20 (message):
  Cannot grep version number: The system cannot find the file specified

I have already install grep (GnuWin32) on my system and add "C:\Program Files (x86)\GnuWin32\bin" on system variable path but it seems like CMake can't find it.
Thank you very much for your time.

aguenette, 2021-02-07:
Hi @avgilias!

Can you try replacing the following lines in the `CMakeLists.txt`

```cmake
execute_process(
	COMMAND grep "NABO_VERSION " nabo/nabo.h
	WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
	RESULT_VARIABLE GREP_VERSION_RESULT
	OUTPUT_VARIABLE PROJECT_VERSION
	OUTPUT_STRIP_TRAILING_WHITESPACE
)

if (NOT GREP_VERSION_RESULT EQUAL 0)
	message(SEND_ERROR "Cannot grep version number: ${GREP_VERSION_RESULT}")
endif ()
```

with these lines instead?

```cmake
file(STRINGS "nabo/nabo.h" PROJECT_VERSION REGEX "#define NABO_VERSION ")

if (NOT PROJECT_VERSION)
	message(SEND_ERROR "Cannot grep version number: ${PROJECT_VERSION}")
endif ()
```

It's the same solution used for `libpointmatcher` to retrieve the version.

avgilias, 2021-02-08:
@aguenette That worked perfectly. Thank you alot for your help and your time

closed 2021-02-08 by avgilias

aguenette, 2021-02-08:
@avgilias you're welcome! 🙂