norlab-ulaval/libpointmatcher #444: Compile Error due to libnabo

maipbui, 2021-02-10:
Hello,

I want to install libpointmatcher version 1.2.4 and followed the instructions but failed to compile. Could you please check the error message and help? Thank you. It happens same error when I tried on Ubuntu 16.04.

Ubuntu: 18.04
64-bit
git version 2.17.1
cmake version 3.10.2
GCC version:
```mb18@mb18-Alienware-15-R3:~/Libraries/libnabo$ gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Boost version: ```Version: 1.65.1.0ubuntu1```
Eigen3: ```Version: 3.3.4-4```
No doxygen is installed
libnabo: install from  [source](https://github.com/ethz-asl/libnabo)

Error message: 
```
mb18@mb18-Alienware-15-R3:~/Libraries/libpointmatcher-1.2.4/build$ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo ${SRC_DIR}
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test FLAG_WEXTRA
-- Performing Test FLAG_WEXTRA - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   filesystem
--   system
--   program_options
--   date_time
--   chrono
--   atomic
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   filesystem
--   system
--   program_options
--   date_time
--   chrono
--   atomic
-- libnabo found, version 1.0.7 (include= libs=)
CMake Warning (dev) at CMakeLists.txt:218 (get_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
[... 1424 lines cut ...]
../libpointmatcher.so.1.2.4: undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/align_sequence.dir/build.make:104: recipe for target 'examples/align_sequence' failed
make[2]: *** [examples/align_sequence] Error 1
CMakeFiles/Makefile2:457: recipe for target 'examples/CMakeFiles/align_sequence.dir/all' failed
make[1]: *** [examples/CMakeFiles/align_sequence.dir/all] Error 2
../libpointmatcher.so.1.2.4: undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
../libpointmatcher.so.1.2.4: undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/build_map.dir/build.make:104: recipe for target 'examples/build_map' failed
make[2]: *** [examples/build_map] Error 1
CMakeFiles/Makefile2:568: recipe for target 'examples/CMakeFiles/build_map.dir/all' failed
make[1]: *** [examples/CMakeFiles/build_map.dir/all] Error 2
[ 89%] Linking CXX executable compute_overlap
../libpointmatcher.so.1.2.4: undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
../libpointmatcher.so.1.2.4: undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/compute_overlap.dir/build.make:104: recipe for target 'examples/compute_overlap' failed
make[2]: *** [examples/compute_overlap] Error 1
CMakeFiles/Makefile2:605: recipe for target 'examples/CMakeFiles/compute_overlap.dir/all' failed
make[1]: *** [examples/CMakeFiles/compute_overlap.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

aguenette, 2021-02-11:
Hi @maipbui! You have two options to be able to compile libpointmatcher:

1. Use version 1.2.4, 1.3.0 or 1.3.1 of libpointmatcher and version 1.0.7 of libnabo, i.e. to check out their respective tag and build the libraries as follows.
2. Use the latest changes of both libpointmatcher and libnabo, i.e. checkout their master branch and build the libraries as follows.

The reason is that the CMakeLists.txt file on the master branch of both libraries is not compatible with older version of the libraries.

maipbui, 2021-02-11:
> Hi @maipbui! You have two options to be able to compile libpointmatcher:
>
> 1. Use version 1.2.4, 1.3.0 or 1.3.1 of libpointmatcher and version 1.0.7 of libnabo, i.e. to check out their respective tag [...]

Hi @aguenette, thanks for the reply. I installed libnabo version 1.0.7 and libpointmatcher 1.2.4 but it failed so what should I do now as your suggestions?

aguenette, 2021-02-11:
I also use Ubuntu 18.04 and successfully build libpointmatcher 1.2.4 using libnabo 1.0.7. So, what I suggest you to do is to start back from the beginning, i.e. uninstalling libnabo and removing the build directory, then redo the steps to build it and install it. Same thing for libpointmatcher, but only remove the build directory.

maipbui, 2021-02-11:
> I'm also using Ubuntu 18.04 and I successfully builded libpointmatcher 1.2.4 using libnabo 1.0.7. So, what I suggest you to do is to start back from the beginning, i.e. uninstalling libnabo and removing the build directory, then redo the steps to build it and install it. Same thing for libpointmatcher, but only by removing the build directory.

Got what you meant, I was able to compile it. Thank you very much!!!

maipbui, 2021-02-15:
Hello @aguenette,

Does package "libpointmatcher_ros" supported on Ubuntu 18.04, ROS Melodic? I tried to install other software but failed to compile because this below error message. I checked this [documentation](https://wiki.ros.org/ethzasl_icp_mapping) and see it is only release with fuerte, groovy, and hydro. Would it be possible to install "libpointmatcher_ros" on Ros Melodic and could you please show me how? Thanks for the help!
```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "libpointmatcher_ros" with any of the following names:

    libpointmatcher_rosConfig.cmake
    libpointmatcher_ros-config.cmake

  Add the installation prefix of "libpointmatcher_ros" to CMAKE_PREFIX_PATH
  or set "libpointmatcher_ros_DIR" to a directory containing one of the above
  files.  If "libpointmatcher_ros" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  map_compression/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!

```

Chick92, 2022-01-27:
Same problem. Trying solution now, but the solution offered is essentially "use an older version", which seems counter productive.

pomerlef, 2022-01-27:
@Chick92 please give us more details about your version and from where you are fetching them? I think there are multiple version of "libpointmatcher_ros" floating around.

We are currently supporting that one:
https://github.com/norlab-ulaval/libpointmatcher_ros

Chick92, 2022-01-27:
Hi @pomerlef, i've just managed to successfully build it using the master branches from libpointmatcher and libnabo. I had always assumed that the master branches would be the default ones pulled, apparently not. 

Thanks!

closed 2022-01-27 by pomerlef

YibinWu, 2022-11-11:
> 

I tried all three branches of libpointmatcher but failed to make it. Then I switched to master and succeed. I used libnabo 1.0.7.

pomerlef, 2022-11-11:
@YibinWu open another issue and follow https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationUbuntu.md to include the basic information.