norlab-ulaval/libpointmatcher #445: Scans misalignment without any movement in the scene

aldoshkind, 2021-02-11:
Hi!
I faced a problem of misalignment of scans without any movement in the scene being scanned. Problem appears in approx 1 of 10 sequential scans.
It is unclear for my why this does happen. Please clarify this behaviour and suggest ways to solve or at least detect such a misalignment if possible.
Scanning device is Realsense L515
libpointmatcher commit is 6cfdc0dc2b36cef61b25318938c31ef2ec4244fd

I set ICP up like this (but default ICP settings are also affected):

```cpp
PM::ICP icp;

icp.transformations.push_back(PM::get().TransformationRegistrar.create("RigidTransformation"));
//icp.readingDataPointsFilters.push_back(PM::get().DataPointsFilterRegistrar.create("RandomSamplingDataPointsFilter"));
icp.referenceDataPointsFilters.push_back(PM::get().DataPointsFilterRegistrar.create("SamplingSurfaceNormalDataPointsFilter"));
icp.outlierFilters.push_back(PM::get().OutlierFilterRegistrar.create("TrimmedDistOutlierFilter"));
icp.matcher = PM::get().MatcherRegistrar.create("KDTreeMatcher");
icp.errorMinimizer = PM::get().ErrorMinimizerRegistrar.create("PointToPointErrorMinimizer");
icp.transformationCheckers.push_back(PM::get().TransformationCheckerRegistrar.create("CounterTransformationChecker"));
icp.transformationCheckers.push_back(PM::get().TransformationCheckerRegistrar.create("DifferentialTransformationChecker"));
icp.inspector = PM::get().InspectorRegistrar.create("NullInspector");
```

Full code and scans are available here: https://yadi.sk/d/Nr6UybS3-8XO5w

pomerlef, 2021-02-11:
Hard to tell as it is heavily related to your application and point clouds you are feeding to ICP.

aldoshkind, 2021-03-09:
Well, data is just scans of a room, hence it contains lots of planar objects (walls, table surfaces, computer monitors).
My assumption is that this environment is a good test case to understand if libpointmatcher is appropriate for my tasks.
I found two remarkable facts:
- my raw data contains outliers (not much, but tangible);
- residual sum (as proposed in docs) is pretty informative: if two scans matched correctly residual sum is less than 1. Otherwise it is greater than 1, sometimes hundreds or thousands.

Please advise how to tweak algorithm parameters, or debug it.

pomerlef, 2021-03-16:
Your best guest is to debug visually using this kind of inspector:

```
inspector:
  VTKFileInspector:
     baseFileName : vissteps
     dumpDataLinks : 1
     dumpReading : 1
     dumpReference : 1
```

You can have a step-by-step process over here: https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ICPIntro.md

aldoshkind, 2021-03-21:
Thanks a lot! This helped to advance further.
I found that problem disappers (on particular pair of scans) if `knn` of `SamplingSurfaceNormalDataPointsFilter` is set to 9 or more.

With default value (7 if I am not mistaken) I have infinite oscillations. Here is pic of rotation and translation on 200 iterations (`knn = 8`):
[image]

With `knn = 9` there are only 3 iterations:
[image]

Now my questions are:
- is this behavoir correct at all? Those oscillations are pretty much scary :)
- how to pick correct value for `knn`? Or maybe another parameters shoud be adjusted?
- if I increase`knn` - does it guarantee stability?
- how to detect if scans are matched correctly? Which metric should I use?

pomerlef, 2021-03-25:
The oscillation is not good. You can have a look in Paraview to the scans and the matching links to understand what cause it. You could also play with the `knn` of the matching function. Unfortunately, to find the right parameters for you application, you will need to try them. Here some place to start:
- [pico_flexx](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/reintegrate/master_into_indigo_devel/ethzasl_icp_mapper/launch/pmd_pico_flexx/icp.yaml)
- [stereo head for Atlas](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/reintegrate/master_into_indigo_devel/ethzasl_icp_mapper/launch/multiSenseSL/icp.yaml)
- [early kinect](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/reintegrate/master_into_indigo_devel/ethzasl_icp_mapper/launch/openni/icp.yaml)

Finding the is the scan aligned properly assume that you know the answer. You can detect few problems by looking at the residual error, but all. There is an ongoing conversation on that over here #443 .