norlab-ulaval/libpointmatcher #451: CMake fails for downstream package

skasperski, 2021-02-23:
Since a recent update my package using pointmatcher fails with a CMake error. I installed pointmatcher via normal "cmake + make install" and use it from my package like this:

```
find_package(libpointmatcher REQUIRED)
add_library(pm_test test.cpp)
target_link_libraries(pm_test PUBLIC pointmatcher)
```
However this fails with:
```
CMake Error in CMakeLists.txt:
  Imported target "pointmatcher" includes non-existent path

    "/usr/local/pointmatcher"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```
Trying to solve this I stumbled across multiple (and different) versions of pointmatcher's cmake-files in these 2 locations:
```
/usr/local/lib/cmake/pointmatcher
-rw-r--r-- 1 root root 3310 Feb 16 16:48 libpointmatcherConfig.cmake
-rw-r--r-- 1 root root  362 Feb 16 16:48 libpointmatcherConfigVersion.cmake
```
and
```
/usr/local/share/libpointmatcher/cmake
-rw-r--r-- 1 root root 3908 Feb 16 16:48 libpointmatcher-config.cmake
-rw-r--r-- 1 root root 3310 Feb 16 16:48 libpointmatcherConfig.cmake
-rw-r--r-- 1 root root  833 Feb 16 16:48 libpointmatcher-config-release.cmake
-rw-r--r-- 1 root root  362 Feb 16 16:48 libpointmatcherConfigVersion.cmake
-rw-r--r-- 1 root root 3137 Feb 16 16:48 yaml-cpp-pm-targets.cmake
-rw-r--r-- 1 root root  826 Feb 16 16:48 yaml-cpp-pm-targets-release.cmake
```
I assume this makes it undefined which version is actually used by CMake's find_package call.

aguenette, 2021-02-24:
Hi @skasperski! Did you find the problem? If not, you can try this instead:

```cmake
find_package(libpointmatcher REQUIRED)
include_directories(${libpointmatcher_INCLUDE_DIRS})
add_library(pm_test test.cpp)
target_link_libraries(pm_test PUBLIC ${libpointmatcher_LIBRARIES})
```

skasperski, 2021-02-25:
This gives me:
```
CMake Error:
  Error evaluating generator expression:

    $<TARGET_FILE:pointmatcher>

  No target "pointmatcher"
```

aguenette, 2021-03-07:
Have you tried to configure your project using a new build directory?

If it does not solve your issue, I would need to know which cmake's version you are using and what is the minimum version that is configured in your project's CMakeLists.txt file. Thanks!

pomerlef, 2021-03-07:
Also, if you're on Ubuntu, give us the output of that script to simplify the process: https://github.com/ethz-asl/libpointmatcher/blob/master/utest/listVersionsUbuntu.sh

skasperski, 2021-03-19:
Sorry for late reply, somehow lost track of this.

I completely rebuild everything several times. My cmake version is 3.10.2 (default on Ubuntu 18.04). In my CMakeLists I have "cmake_minimum_required(VERSION 3.0)". 

Output of the script above:
```
Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 18.04.5 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
git: 		| git version 2.17.1
cmake: 		| cmake version 3.10.2 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: 		| Version: 1.65.1.0ubuntu1
eigen3: 	| Version: 3.3.4-4
doxygen: 	| Version: 1.8.13-10
```

skasperski, 2021-05-28:
I now also tried building this inside a catkin workspace, which gives me the same error.