norlab-ulaval/libpointmatcher #455: Working in Debug mode

danielaDESM, 2021-03-21:
Hi!

I have installed successfully the library. I am creating a new project and I am running some of the sample codes that you have already available. Everything works fine, but only when I build my project in Release mode, when I change to Debug mode I get the following errors:
[image]
Is it possible to work in Debug mode and do you know how?

pomerlef, 2021-03-21:
Are you on Windows? I'm not familiar with building on it, but it looks like there is a mismatch with something else that was not build in `Debug`. Maybe you need to build libnabo in debug too. It's not the case on Ubuntu.

Also, I tend to use more `RelWithDebInfo`.

danielaDESM, 2021-03-21:
Hi,I am in Windows.

danielaDESM, 2021-03-21:
Thanks @pomerlef !
So the Debug version still does not work, but I tried adding as well RelWithDebInfo in the build of libnabo and then in the build of libpointmatcher. They built successfully and now I am able to debug in my own project.

closed 2021-03-21 by danielaDESM