norlab-ulaval/libpointmatcher #458: undefined reference error in compiling

Seikegn, 2021-04-06:
Dear developers,
    I want to thank you for creating this comprehensive point cloud toolkit, and some of the filters are exactly what I need now. I'm literally new to C++ language and not very familiar with Linux, so it took me  ~20 hrs to set up eigen, boost and libnabo and "successfully" make install libpointmatcher through trials and errors. However, when I try compiling examples/demo_cmake/convert.cpp, the following error messages were thrown:

/tmp/ccokO4LU.o: In function `main':
convert.cpp:(.text+0x163): undefined reference to `PointMatcher<float>::get()'
convert.cpp:(.text+0x202): undefined reference to `PointMatcherSupport::setLogger(std::shared_ptr<PointMatcherSupport::Logger>)'
convert.cpp:(.text+0x28d): undefined reference to `PointMatcher<float>::DataPoints::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert.cpp:(.text+0x360): undefined reference to `PointMatcher<float>::DataPointsFilters::DataPointsFilters(std::istream&)'
convert.cpp:(.text+0x379): undefined reference to `PointMatcher<float>::DataPointsFilters::apply(PointMatcher<float>::DataPoints&)'
convert.cpp:(.text+0x3fa): undefined reference to `PointMatcher<float>::DataPoints::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) const'
/tmp/ccokO4LU.o: In function `PointMatcherSupport::Registrar<PointMatcherSupport::Logger>::getDescriptor(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
convert.cpp:(.text._ZNK19PointMatcherSupport9RegistrarINS_6LoggerEE13getDescriptorERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK19PointMatcherSupport9RegistrarINS_6LoggerEE13getDescriptorERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x11c): undefined reference to `PointMatcherSupport::InvalidElement::InvalidElement(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status

    Which I don't even know how to debug and almost destroyed my will to make use of libpointmatcher as I have spend too much time on merely setting it up. So I'm giving myself a last chance to open an issue here, hoping to get some guidance.

    Related environment is: 
    boost 1.70.0 (non-conda)
    eigen 3.3.9
    libnabo 1.07
    g++ 9.3.0 and gcc 9.3.0 (non-conda)
    cmake 3.18.4 (conda)
    make 3.82
    system (on HPC cluster with slurm) 
                  LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop- 
                  4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1- 
                  amd64:printing-4.1-noarch
                  Distributor ID:	CentOS
                  Description:	CentOS Linux release 7.6.1810 (Core) 
                  Release:	7.6.1810
                  Codename:	Core   
    I have put boost, eigen, libnabo and libpointmatcher into CPATH.

    What I have tried:
    1. The head part of examples/demo_cmake/convert.cpp can be compiled successfully
      #include "pointmatcher/PointMatcher.h"
      #include <cassert>
      #include <iostream>
      #include <fstream>
      
      using namespace std;
      using namespace PointMatcherSupport;
      
      typedef PointMatcher<float> PM;
      typedef PM::DataPoints DP;

      while any more than that would cause error message.

    2. Compiling libnabo/examples/usage.cpp throws the following error:
        /tmp/cc1taue8.o: In function `main':
        usage.cpp:(.text+0x94): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::createKDTreeLinearHeap(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, unsigned int, Nabo::Parameters const&)'
        collect2: error: ld returned 1 exit status
       
        Compiling libnabo/examples/trivial.cpp throws the following error:
        /tmp/ccXKCvIU.o: In function `main':
        trivial.cpp:(.text+0x89): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::createKDTreeLinearHeap(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, unsigned int, Nabo::Parameters const&)'
        trivial.cpp:(.text+0x10e): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::knn(Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, Eigen::Matrix<int, -1, 1, 0, -1, 1>&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&, int, float, unsigned int, float) const'
        collect2: error: ld returned 1 exit status
        
        Please allow me to put it here as that might be the cause of libpointmatcher failure and the repo of libnabo is short of maintenance.
    
    3. Changing boost 1.70.0 to 1.74.0 or 1.75.0 would cause some "undefined reference" error during making libpointmatcher, let alone to say using it.

    Please forgive my possible informality in describing the situation. If any other information should be required, please just ask and I'll add it as long as I'm able to.

    Bests.

pomerlef, 2021-04-06:
@aguenette could you have a look?

aguenette, 2021-04-06:
Hi @Seikegn! Can you add below's lines to the `demo_cmake` `CMakeLists.txt` file after the `find_package(libpointmatcher REQUIRED)` line and give me the output please?

```cmake
message("libpointmatcher_INCLUDE_DIRS: ${libpointmatcher_INCLUDE_DIRS}\n")
message("libpointmatcher_LIBRARIES: ${libpointmatcher_LIBRARIES}\n")
```

Seikegn, 2021-04-07:
Thanks a lot! @aguenette @pomerlef 
@aguenette I assume you asked me to add the two lines and then run `cmake .`. Below is the direct output with cmake 3.18.4.

    -- The C compiler identification is GNU 9.3.0
    -- The CXX compiler identification is GNU 9.3.0
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working C compiler: //appsnew/usr/gcc/gcc-9.3.0/bin/gcc - skipped
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Check for working CXX compiler: /appsnew/usr/gcc/gcc-9.3.0/bin/g++ - skipped
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    CMake Warning (dev) at /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib/python3.7/site-packages/cmake/data/share/cmake-3.18/Modules/FindBoost.cmake:1922 (if):
      Policy CMP0054 is not set: Only interpret if() arguments as variables or
      keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
      details.  Use the cmake_policy command to set the policy and suppress this
      warning.
    
      Quoted variables like "thread" will no longer be dereferenced when the
      policy is set to NEW.  Since the policy is not set the OLD behavior will be
      used.
    Call Stack (most recent call first):
      /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/built/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:10 (find_package)
      CMakeLists.txt:10 (find_package)
    This warning is for project developers.  Use -Wno-dev to suppress it.
    
    CMake Warning (dev) at /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib/python3.7/site-packages/cmake/data/share/cmake-3.18/Modules/FindBoost.cmake:2053 (if):
      Policy CMP0054 is not set: Only interpret if() arguments as variables or
      keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
      details.  Use the cmake_policy command to set the policy and suppress this
      warning.
    
      Quoted variables like "thread" will no longer be dereferenced when the
      policy is set to NEW.  Since the policy is not set the OLD behavior will be
      used.
    Call Stack (most recent call first):
      /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/built/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:10 (find_package)
      CMakeLists.txt:10 (find_package)
    This warning is for project developers.  Use -Wno-dev to suppress it.
    
    CMake Warning (dev) at /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib/python3.7/site-packages/cmake/data/share/cmake-3.18/Modules/FindBoost.cmake:2108 (if):
      Policy CMP0054 is not set: Only interpret if() arguments as variables or
      keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
      details.  Use the cmake_policy command to set the policy and suppress this
      warning.
    
      Quoted variables like "thread" will no longer be dereferenced when the
      policy is set to NEW.  Since the policy is not set the OLD behavior will be
      used.
    Call Stack (most recent call first):
      /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/built/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:10 (find_package)
      CMakeLists.txt:10 (find_package)
    This warning is for project developers.  Use -Wno-dev to suppress it.
    
    CMake Warning (dev) at /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib/python3.7/site-packages/cmake/data/share/cmake-3.18/Modules/FindBoost.cmake:2346 (if):
      Policy CMP0054 is not set: Only interpret if() arguments as variables or
      keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
      details.  Use the cmake_policy command to set the policy and suppress this
      warning.
    
      Quoted variables like "thread" will no longer be dereferenced when the
      policy is set to NEW.  Since the policy is not set the OLD behavior will be
      used.
    Call Stack (most recent call first):
      /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/built/share/libpointmatcher/cmake/libpointmatcherConfig.cmake:10 (find_package)
      CMakeLists.txt:10 (find_package)
    This warning is for project developers.  Use -Wno-dev to suppress it.
[... 437 lines cut ...]
        arg [-rpath-link] ==> ignore
        arg [/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib] ==> ignore
        arg [CMakeFiles/cmTC_7293b.dir/CMakeCXXCompilerABI.cpp.o] ==> ignore
        arg [-lstdc++] ==> lib [stdc++]
        arg [-lm] ==> lib [m]
        arg [-lgcc_s] ==> lib [gcc_s]
        arg [-lgcc] ==> lib [gcc]
        arg [-lc] ==> lib [c]
        arg [-lgcc_s] ==> lib [gcc_s]
        arg [-lgcc] ==> lib [gcc]
        arg [/appsnew/usr/gcc/gcc-9.3.0/lib/gcc/x86_64-pc-linux-gnu/9.3.0/crtend.o] ==> ignore
        arg [/lib/../lib64/crtn.o] ==> ignore
      collapse library dir [/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib] ==> [/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib]
      collapse library dir [/appsnew/usr/gcc/gcc-9.3.0/lib/gcc/x86_64-pc-linux-gnu/9.3.0] ==> [/appsnew/usr/gcc/gcc-9.3.0/lib/gcc/x86_64-pc-linux-gnu/9.3.0]
      collapse library dir [/appsnew/usr/gcc/gcc-9.3.0/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../lib64] ==> [/appsnew/usr/gcc/gcc-9.3.0/lib64]
[... cut ...]
      collapse library dir [/appsnew/usr/gcc/gcc-9.3.0/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../..] ==> [/appsnew/usr/gcc/gcc-9.3.0/lib]
      implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
      implicit dirs: [/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/Anaconda3/envs/PCT/lib;/appsnew/usr/gcc/gcc-9.3.0/lib/gcc/x86_64-pc-linux-gnu/9.3.0;/appsnew/usr/gcc/gcc-9.3.0/lib64;/lib64;/usr/lib64;/appsnew/usr/gcc/gcc-9.3.0/lib]
      implicit fwks: []

So sorry to distract you with a log this lengthy, but I'm not sure which part might be pertinent.

Please let me know if any other information should be needed.

aguenette, 2021-04-07:
> @aguenette I assume you asked me to add the two lines and then run cmake .. Below is the direct output with cmake 3.18.4.

Yes, that's what I meant, thanks for the feedback!

---

It looks like your `libpointmatcher_INCLUDE_DIRS` variable is different from what I get on Ubuntu:

`libpointmatcher_INCLUDE_DIRS: /usr/local/include;/usr/include/eigen3`

Yours is:

`libpointmatcher_INCLUDE_DIRS: /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/built/include;/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/eigen-3.3.9;/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/pointmatcher;/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/PUBLIC;/lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libnabo/built/inclue`

For example, when I build `libpointmatcher` or `libnabo` on Ubuntu, there's no `include` directory in my `build` directory (plus there's a typo in your libnabo's one). So it might be that... or not.

Also, I thought that it could be your `cmake 3.18.4` that you are using, you get policy errors that I don't get even if I use `cmake 3.20.0`. Maybe, you could try to install it from `CentOS` package manager instead of anaconda?

You also mentioned that you struggled installing the library and its dependencies, can you tell me what you had to do in order to build the library? Like, have you installed `libnabo` and `libpointmatcher`? Did you have to manually specified the path to both libraries? Have you change the `CMakeLists.txt. file in the demo project? That kind of stuff.

It also looks like you're doing an in-source build for demo project:

`-- Build files have been written to: /lustre2/jdhan_pkuhpc/yangxinyu/SoftRepo/libpointmatcher/examples/demo_cmake`

I recommend you to create a `build` directory and do an out-of-source build like you did for `libpointmatcher` and `libnabo`. Doing so, you will be able to remove the directory and start with a new configuration if something goes wrong with the building process.

Finally, if you have installed `libointmatcher` and `libnabo` on your system, I suggest you to uninstall them and try to rebuild them with another version of `cmake` and maybe use `cmake-gui` to set some variable a little more easily and don't forget to remove the `build` directory and create a new one. I hope this will help you to successfully install and use `libpointmatcher`. If you have any questions about what I said, don't hesitate.

brunoeducsantos, 2021-04-08:
I had an error compiling `python bind` 
```
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:804: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/octree_grid.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:336: python/CMakeFiles/pypointmatcher.dir/pointmatcher/transformations.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:778: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/normal_space.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:596: python/CMakeFiles/pypointmatcher.dir/errorminimizers/point_to_point_similarity.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:700: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/gestalt.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:765: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/max_quantile_on_axis.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:76: python/CMakeFiles/pypointmatcher.dir/pointmatcher/data_points_filter.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:817: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/orient_normals.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:518: python/CMakeFiles/pypointmatcher.dir/pointmatchersupport/registrars/transformation_checker_registrar.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:401: python/CMakeFiles/pypointmatcher.dir/pointmatchersupport/logger_impl.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:856: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/remove_sensor_bias.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:128: python/CMakeFiles/pypointmatcher.dir/pointmatcher/icp_chain_base.cpp.o] Error 1
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:89: python/CMakeFiles/pypointmatcher.dir/pointmatcher/data_points_filters.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:310: python/CMakeFiles/pypointmatcher.dir/pointmatcher/point_matcher.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:570: python/CMakeFiles/pypointmatcher.dir/errorminimizers/point_to_plane_with_cov.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:713: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/identity.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:609: python/CMakeFiles/pypointmatcher.dir/errorminimizers/point_to_point_with_cov.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:531: python/CMakeFiles/pypointmatcher.dir/pointmatchersupport/registrar.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:271: python/CMakeFiles/pypointmatcher.dir/pointmatcher/matches.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:726: python/CMakeFiles/pypointmatcher.dir/datapointsfilters/incidence_angle.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:986: python/CMakeFiles/pypointmatcher.dir/pypoint_matcher.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:141: python/CMakeFiles/pypointmatcher.dir/pointmatcher/icp_sequence.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[... 18 lines cut ...]
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:414: python/CMakeFiles/pypointmatcher.dir/pointmatchersupport/parametrizable.cpp.o] Error 1
[... cut ...]
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:232: python/CMakeFiles/pypointmatcher.dir/pointmatcher/inspector.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [python/CMakeFiles/pypointmatcher.dir/build.make:297: python/CMakeFiles/pypointmatcher.dir/pointmatcher/outlier_filters.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:574: python/CMakeFiles/pypointmatcher.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:581: python/CMakeFiles/pypointmatcher.dir/rule] Error 2

```
Do I need to install first the `cpp` version?
Thanks