norlab-ulaval/libpointmatcher #461: Distance minimization:SVD or LSQ

brunoeducsantos, 2021-04-16:
Hi,
Thanks for the amazing work.

Does this libpointmatcher use SVD approach or robust least squares for minimizing squared error ?

Regards,

Bruno

pomerlef, 2021-04-16:
It depends of the error minimizer and outlier rejection used in your configuration. Only the point-to-point error used SVD, which can be robustified with a weighted cost function or a binary outlier rejection of your choice. Libpm is not a single registration solution, but a library of modules to build you own solution. If it can help, you can read this survey for more information:

F. Pomerleau, F. Colas and R. Siegwart (2015), "A Review of Point Cloud Registration Algorithms for Mobile Robotics", Foundations and Trends® in Robotics: Vol. 4: No. 1, pp 1-104. https://doi.org/10.1561/2300000035

brunoeducsantos, 2021-04-16:
Thanks for your insight .

closed 2021-04-16 by brunoeducsantos