norlab-ulaval/libpointmatcher #475: Example of configuring parameter of RobustOutlierFilter with yaml

dasisttao, 2021-11-11:
I have found that the RobustOutlierFilter like cauchy has "tuning" parameter. Is there a example for me to config this tuning parameter quickly with yaml file?  Thanks!

My config right now is 
```
    outlierFilters:
      - RobustOutlierFilter:
         welsch
```

pomerlef, 2021-11-11:
You can always have more information using the executable:

```
./pmicp -l
```

which would give
[...]
```
RobustOutlierFilter
Robust weight function. 8 robust functions to choose from (Cauchy, Welsch, Switchable Constraint, Geman-McClure, Tukey, Huber, L1 and Student). All the functions are M-Estimator (\cite{RobustWeightFunctions}) except L1 and Student.
- robustFct (default: cauchy) - Type of robust function used. Available fct: 'cauchy', 'welsch', 'sc'(aka Switchable-Constraint), 'gm' (aka Geman-McClure), 'tukey', 'huber' and 'L1'. (Default: cauchy)
- tuning (default: 1.0) - Tuning parameter used to limit the influence of outliers.If the 'scaleEstimator' is 'mad' or 'none', this parameter acts as the tuning parameter.If the 'scaleEstimator' is 'berg' this parameter acts as the target scale (σ*). - min: 0.0000001 - max: inf
- scaleEstimator (default: mad) - The scale estimator is used to convert the error distance into a Mahalanobis distance. 3 estimators are available: 'none': no estimator (scale = 1), 'mad': use the median of absolute deviation (a kind of robust standard deviation), 'berg': an iterative exponentially decreasing estimator
- nbIterationForScale (default: 0) - For how many iteration the 'scaleEstimator' is recalculated. After 'nbIterationForScale' iteration the previous scale is kept. A nbIterationForScale==0 means that the estiamtor is recalculated at each iteration. - min: 0 - max: 100
- distanceType (default: point2point) - Type of error distance used, either point to point ('point2point') or point to plane('point2plane'). Point to point gives better result normally.
- approximation (default: inf) - If the matched distance is larger than this threshold, its weight will be forced to zero. This can save computation as zero values are not minimized. If set to inf (default value), no approximation is done. The unit of this parameter is the same as the distance used, typically meters. - min: 0.0 - max: inf
```
[...] 

In your case, it should be something like (not tested):
```
outlierFilters:
  - RobustOutlierFilter:
        robustFct: welsch
        tuning: 6.3
```

Be careful about spacing in your yaml file

A general list of robust functions can be found here: https://github.com/ethz-asl/libpointmatcher/blob/17aa1c10f39c90c896333a5d1750e34da7b43ae8/doc/OutlierFiltersFamilies.md

Don't hesitate to do a PR for the documentation. If you where missing details, you're most probably not the only one.

Cheers!

dasisttao, 2021-11-11:
Thank you! I would like to do a PR when I can add some tested details in the doc.  I close this issue now.

closed 2021-11-11 by dasisttao