norlab-ulaval/libpointmatcher #478: pypointmatcher.cpython-38-x86_64-linux-gnu.so: undefined symbol

venustar2015, 2021-11-22:
I am trying to use libpointmatcher with python, but when I import the module, it appears the error 'undefined symbol'. Here is the detailed error information:

```
Python 3.8.8 (default, Apr 13 2021, 19:58:26) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pypointmatcher
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /home/dfcv/anaconda3/lib/python3.8/site-packages/pypointmatcher.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN14PointMatcherIOIfE16plyPropTypeValidERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
```

Here is the other information:

`(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ lsb_release -r
Release:        16.04
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ getconf LONG_BIT
64
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ git --version
git version 2.7.4
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ cmake --version
cmake version 3.5.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ dpkg -s libboost-dev | grep Version
Version: 1.58.0.1ubuntu1
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ dpkg -s libeigen3-dev | grep Version
Version: 3.3~beta1-2
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/examples/python$ dpkg -s doxygen | grep Version
Version: 1.8.11-1ubuntu0.1`

pomerlef, 2021-11-22:
@aguenette any thoughts?

aguenette, 2021-11-23:
@venustar2015 Which version of pybind11 did you use? I've compiled the bindings with python3.8 and everything is working. It looks like it cannot find the method `plyPropTypeValid` from `IO.cpp`. Can you take a screenshot of cmake-gui? If you check the `grouped` and `advanced` checkboxes, you should get something like this:

[image]

You can expand only the `Ungrouped entries` and the `PYTHON` groups. Thanks!

venustar2015, 2021-11-24:
> @venustar2015 Which version of pybind11 did you use? I've compiled the bindings with python3.8 and everything is working. It looks like it cannot find the method `plyPropTypeValid` from `IO.cpp`. [...]

Dear Aguenette, 
it's very glad to hear from you. The version of pybind11 is v2.5.0 as it's described in tutorials, but I don't know how to check it.
I get the screenshot for cmake-gui, but cannot find  the "pybind11" entries and "PYTHON" grouped as your screenshot. What is the possible cause for it?
Here is the screenshot of cmake-gui:

[image]

Hope you have a nice day. Thanks!

aguenette, 2021-11-27:
@venustar2015 I don't know why these two groups aren't there. To check for the pybind11 version, it should appear when you configure your build directory with cmake, i.e. running `cmake ..` in the build directory. Could you copy and paste here the output after running this command? I'm also curious to see what's in the `BUILD` group from cmake-gui. Thanks!

venustar2015, 2021-11-29:
Dear Aguenette,

Thanks very much for your reply. Here is the info in `pybind11/build` directory foy `cmake ..`

```
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/pybind11/build$ cmake ..
-- Building tests with Eigen v3.2.92
-- Boost version: 1.58.0
-- Catch not detected. Interpreter tests will be skipped. Install Catch headers manually or use `cmake -DDOWNLOAD_CATCH=1` to fetch them automatically.
-- pybind11 v2.5.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dfcv/venus/code/pybind11/build
```

Here's the info in `libpointmatcher/build` directory for `cmake ..`
```
(base) dfcv@dfcv-ThinkCentre-M720t-N000:~/venus/code/libpointmatcher/build$ cmake ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   filesystem
--   system
--   program_options
--   date_time
--   chrono
--   atomic
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   filesystem
--   system
--   program_options
--   date_time
--   chrono
--   atomic
-- libnabo found, version 1.0.7 (include= libs=)
-- OpenMP found, parallel computer enabled
-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
-- API Documentation (doxygen): disabled
-- The Python module will be install at this location : /home/dfcv/anaconda3/lib/python3.8/site-packages
-- pybind11 v2.5.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dfcv/venus/code/libpointmatcher/build
```

Here is the cmake-gui info. After I check BUILD_PYTHON_MOUDLE and uncheck BUILD_TESTS in BUILD group, and configure again, the PYTHON group and pybind11 info appears, just like below.
[image]

However, the error `undefined symbol` still exists.