norlab-ulaval/libpointmatcher #481: received signal SIGSEGV, Segmentation fault

zhangyuan12121, 2021-12-01:
Ｔhank you very much for your work. I have this problem when using libpointmatcher. I spent a long time to solve this problem, but there is still no result.
`Thread 1 "localizer_node" received signal SIGSEGV, Segmentation fault.
0x00007ffff7bb0515 in void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<float, -1, -1, 0, -1, -1>, Eigen::Matrix<float, -1, -1, 0, -1, -1>, Eigen::internal::assign_op<float, float> >(Eigen::Matrix<float, -1, -1, 0, -1, -1>&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, Eigen::internal::assign_op<float, float> const&)
    () from /home/yuan/icp_localization_ws/devel/.private/icp_localization/lib/libicp_localization.so
(gdb) [icp_node-2] killing on exit
Quit
(gdb) bt
#0  0x00007ffff7bb0515 in void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<float, -1, -1, 0, -1, -1>, Eigen::Matrix<float, -1, -1, 0, -1, -1>, Eigen::internal::assign_op<float, float> >(Eigen::Matrix<float, -1, -1, 0, -1, -1>&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, Eigen::internal::assign_op<float, float> const&) () from /home/yuan/icp_localization_ws/devel/.private/icp_localization/lib/libicp_localization.so
#1  0x00007ffff5947b15 in SamplingSurfaceNormalDataPointsFilter<float>::fuseRange(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int) const ()
   from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#2  0x00007ffff5949e15 in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#3  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#4  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#5  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#6  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#7  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#8  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#9  0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#10 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#11 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#12 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#13 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#14 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#15 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#16 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#17 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#18 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#19 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#20 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
#21 0x00007ffff594a2ba in SamplingSurfaceNormalDataPointsFilter<float>::buildNew(SamplingSurfaceNormalDataPointsFilter<float>::BuildData&, int, int, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&, Eigen::Matrix<float, -1, 1, 0, -1, 1>&&) const () from /home/yuan/icp_localization_ws/devel/.private/libpointmatcher/lib/libpointmatcher.so
`

pomerlef, 2021-12-01:
Could you report more information on your system. Some thing along: https://github.com/ethz-asl/libpointmatcher/blob/master/utest/listVersionsUbuntu.sh

zhangyuan12121, 2021-12-01:
Shocked by your efficiency, this is the information of the system：

Name 		| Version
----------------|-------------------------------
./listVersionsUbuntu.sh: line 7: lsb_release: command not found
ubuntu: 	|
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 7.5.0-3ubuntu1~16.04) 7.5.0
git: 		| git version 2.7.4
cmake: 		| cmake version 3.20.5 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: 		| Version: 1.58.0.1ubuntu1
eigen3: 	| Version: 3.3~beta1-2
doxygen: 	| Version: 1.8.11-1ubuntu0.1

pomerlef, 2021-12-02:
and do you have a minimum code example that trigger this segfault? I don't understand why you have a long list of `buildNew(SamplingSurfaceNormalDataPointsFilter [...]` in the backtrace.