norlab-ulaval/libpointmatcher #484: Tree detection, clustering and its removal

tejalbarnwal, 2021-12-11:
How can I perform tree detection, clustering and removal of it using libpointmatcher filters? Are there any other methods to do the same?

pomerlef, 2021-12-13:
You can have a look at these publications: https://norlab.ulaval.ca/publications/forest-mapping/

There is no such capability in libpointmatcher, but any additional filters are welcome. For the publications above, [DBSCAN](https://en.wikipedia.org/wiki/DBSCAN) was tested, but triggered too many false positive for the segmentation. We manually segmented trees for our experiments. The dataset is available if you want to evaluate your solution: https://norlab.ulaval.ca/research/montmorencydataset/

Maybe @jefft255 have something to add?

tejalbarnwal, 2021-12-17:
Thank you, @pomerlef I will definitely take a look at this!

closed 2021-12-17 by pomerlef