norlab-ulaval/libpointmatcher #500: The same configuration but the different results.

ljjTYJR, 2022-09-12:
Hello,

Recently, I have been trying to test some registration methods.

But I got some different results from `PCL-ICP` and `LibpointMatcher`:

- For `PCL-ICP` implementation, I run the same "chain" similar to the paper (`Filter` + `randomSampling` + `Trimmed(75%)` + The same `converge criteria`), and then I got the following results (Only the easy pose results.):

[image]

- For `LibpointMatcher`, I run the same chain, but get better results:
[image]

Do you know if there is any difference between these two implementations?

Best,
Shuo.

pomerlef, 2022-09-12:
Hi Shuo,

Do they both have the same error minimizer (I assume point-to-plane)? This module has the highest chance of being implemented differently. Also, for stochastic filters (e.g., random filters), you will need to run multiple time the same experiment to get a stable overview. 

I hope this help. Cheers!

ljjTYJR, 2022-09-12:
1. They both use the `point-to-point` error metric.
2. Yeah, I think the `random sampling` may have the influence because I found that even for the same two scans, the registration result may vary largely. I will check it.

Thank you!

ljjTYJR, 2022-10-22:
Hello, I have another question.
In the paper **Comparing ICP variants on real-world data sets**, random downsampling is used.

I wonder why choosing random downsampling. From my experiments, using voxel downsampling seems better than random downsampling.

pomerlef, 2022-10-23:
The article date from 9 years ago, so computation power was on the edge. I'm pretty sure that voxel is a bit better, but comes with more computation cost (random downsampling is the fastest to compute). @boxanm might have something to add on this as he compared many filters recently.

closed 2022-12-06 by pomerlef