norlab-ulaval/libpointmatcher #507: Question about Center Of Mass in Point-To-Plane ICP

YoshuaNava, 2023-03-03:
Hi,
I'm debugging some strange behavior of Point-To-Plane ICP, which seems to work way better 

1. When point clouds are close to the origin, and
2. When zeroing the Center Of Mass of the Reference, so that it is not removed,

This increases the basin of convergence for rotations, and I can get up to 30-40° pure rotation estimations in one ICP call.

I wanted to ask if there are some references on the procedure to remove the CoM. I checked different point cloud libraries and papers, and everyone seems to handle this aspect differently.

Best,
Yoshua Nava

pomerlef, 2023-03-03:
Hi @YoshuaNava ,

1. I'm not sure what you mean by 'procedure'. Removing the center of mass is simply removing the mean of the coordinates. As for why we do it, it's because at some time I was applying ICP to match a trajectory with GPS coordinates. These GPS coordinates were expressed with very large numbers, which were not registering well. If you think about it for a point cloud very far from its origin, there is no difference between a small translation and a small translation to resolve the error. That being said, I would like to know what the other libraries are doing.
2. Substracting the mean is part of the classic solution for minimizing point-to-point error from [Arun et al., 1987],(see https://hal.science/hal-01178661/document, P. 43).
3. Keep on the good work ;) I'm not sure if there is something to follow up, maybe adding better comments in the source or adding to the documentation.

Cheers!

YoshuaNava, 2023-03-03:
Hi @pomerlef,

Thank you for your prompt reply :)

My motivation for digging into this topic is because I started noticing that libpointmatcher's Point-To-Plane ICP implementation is quite sensitive to the position of the map origin relative to where we are localizing.

During my investigations I implemented a **rotation-only** version of Point-To-Plane ICP, and I experienced first-hand how, the closer to the origin (and the lower the norm of the reference CoM), the more accurate the rotations were estimated, and the larger rotations it could handle. Furthermore, I also observed how, the farther we are from the origin, the higher the likelihood that the estimated rotation affects the translation component in an unexpected way.

Said differently, the CoM computation could be reducing the basin of convergence of P2P ICP the farther away from the origin the CoM is.

> That being said, I would like to know what the other libraries are doing.

Sure thing 🖖 

I made a short summary of different approaches, from libraries and papers.

# Libraries

In this section I'll include screenshots of the code and links to the code block, because GitHub does not support rendering [Permalinks ](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-a-permanent-link-to-a-code-snippet) as code quotations across different repos.

## PCL ⛔

The computation of the **Point-To-Plane** Error does not consider the CoM:

[image]

[Link to code](https://github.com/PointCloudLibrary/pcl/blob/master/registration/include/pcl/registration/impl/transformation_estimation_point_to_plane_lls.hpp#L202-L204)

## Open3D ⛔ 

The computation of the **Point-To-Plane** Error does not consider the CoM:

[image]

## KissICP ⛔ 

The computation of the **Point-To-Point** Error does not consider the CoM:

[image]

[Link to code](https://github.com/PRBonn/kiss-icp/blob/278891d7016cb5096684584420b9cb71e199b329/cpp/kiss_icp/core/Registration.cpp#L69)

## OpenCV Surface Matching module ✅ 

The computation of the **Point-To-Plane** Error considers the CoM.

Steps followed:

1. Compute the CoM of both the reading and reference point clouds,
2. Compute the average of both CoMs,
3. Subtract the average CoM from both the reading and reference point clouds,
4. Subtract the average CoM from the translation component of the ICP-corrected pose,

[image]

[image]

[Link to code from steps 1-3](https://github.com/opencv/opencv_contrib/blob/4.x/modules/surface_matching/src/icp.cpp#L255-L260)

[Link to code from step 4](https://github.com/opencv/opencv_contrib/blob/4.x/modules/surface_matching/src/icp.cpp#L458-L462)

## Cilantro ✅ 

The computation of the **Point-To-Plane** Error considers the CoM.

Steps followed:

1. Compute the CoM of both the reading (`src_mean`) and reference (`dst_mean`) point clouds,
2. Transform the reading point cloud to our "initial guess on its pose relative to the reference frame" (`src_normals_trans_`),
3. Estimate a transform between point clouds, which includes transforming the reading CoM to the reference frame (with the "initial guess on the reading pose in the reference frame"),
[... 65 lines cut ...]
⚠️ **Question - Potential bug** ⚠️ 

Before calling the Point-To-Point error minimizer, we remove the CoM from the reference point cloud:

https://github.com/ethz-asl/libpointmatcher/blob/e45710b421f757d7eadbf03cd93cb12e698098f8/pointmatcher/ICP.cpp#L292-L294

Is the mean of the reference being removed twice?

### Point-To-Plane ICP ⛔ 

The computation of the **Point-To-Plane** Error does not explicitly consider the CoM.

We implement subtraction of the CoM in the `ICP` class, but not in the Point-To-Plane error minimizer:

https://github.com/ethz-asl/libpointmatcher/blob/e45710b421f757d7eadbf03cd93cb12e698098f8/pointmatcher/ICP.cpp#L292-L294

⚠️ **Question - Potential bug** ⚠️ 

We subtract the reference CoM from the reference point cloud, and we transform the initial guess into the reference CoM frame. But we don't consider the reading CoM at all.

Should the reading CoM be considered more actively?

---

Note: I used the tool [UpMath](https://upmath.me/) to format LaTeX for displaying in Markdown

YoshuaNava, 2023-03-20:
@pomerlef I implemented unit tests for ICP conditioning, as well as a fix.

In order to test the conditioning of the registration pipeline I put together new code, and I would like to submit a PR. To do that, I would like to know your opinion on where we should store new functionality. I follow the practice of 'shifting discussions to the left' (discussing design before submitting PRs, the 'what'), so that reviews can focus on the implementation (the 'how')

Some of the utilities I implemented are targeted to:
* Filesystem: the unit tests allow you to save the test data to disk, to enable debugging if a test fails (all functions relying on `boost::filesystem`)
* Geometry: comparing transformations, generating random translation/rotations, and converting radians/degrees (all functions relying on `Eigen` and `cmath`),
* Gtest: fetching the name of the current test, recording the random seed as a [GTest Property](http://google.github.io/googletest/reference/testing.html#Test::RecordProperty),
* Registration benchmarking: structures for the input and output of registration tests, as well as for measuring the error.

Would you like me to add these utilities? If yes, should I do so under the `utest` directory, or make them generally available under the `pointmatcher` directory & namespace?

Thank you in advance.