norlab-ulaval/libpointmatcher #511: Supporting a newer version of C++

YoshuaNava, 2023-03-29:
At the moment libpointmatcher supports C++11. This version of the language was a big leap, but there are many new features in C++14, 17 and 20, that could be of interest to us.

For example:
* The new memory-aligned operator, which enables the memory arrangement of Eigen matrices to be considered at construction time by default -> https://www.cppstories.com/2019/08/newnew-align/
* Optional, filesystem, promise and future, have made it into the language,
* Multi-returns with structured bindings -> https://en.cppreference.com/w/cpp/language/structured_binding

If everyone is in I would be happy to submit a PR and help adapting any code. I did that for our internal libpointmatcher fork, and it was quite a nice and cheap upgrade :+1:

josefgraus, 2023-04-20:
This is a roadblock for me in using this library. I'm currently using the cmake generator for Visual Studio 17 2022 with dependencies being gathered, as much as possible via Conan v2.0 from ConanCenter. I'm able to satisfy all the external dependencies specified in the Windows Compilation docs (except for grep but that is resolved with no issue via the installer). 

I'm sure you're already aware, but it appears that the main issue holding back compiling for C++20 is the version of yaml-cpp expected. I hacked a newer version (yaml-cpp 0.7.0) and made fairly decent progress getting libpointmatcher nearly compiled but there are a number of spots where the interface for yaml-cpp has changed since the point where they removed `std::auto_ptr` from their codebase to support newer versions of C++. The issues I've discovered so far include:

1. YAML text files are being loaded with a deprecated interface: https://stackoverflow.com/a/41430777/1279176
2. YAML node traversal uses iterators in a way that is no longer supported
3. As far as I can tell, there is no longer a YAML_PM namespace, so all the conditional compilation directives differentiating system vs packaged yaml-cpp can be removed.

I'm uncertain if that's *all* the issues, but those are the biggest ones I ran into before pausing. 

As an aside, the version of gtest included in `contrib/` was also giving me trouble so for purposes of my attempts to get something built I kept the tests disabled and just didn't add the gtest subdirectory via cmake.

Looking forward to updates if you have the time, and I hope it's a fairly easy task? Famous last words haha.

boxanm, 2023-11-17:
I recently merged https://github.com/norlab-ulaval/libpointmatcher/pull/450, which removes the dependency on the old `yaml-cpp `API. This puts us one step closer to supporting a more modern C++ version. @josefgraus , can you expand on what were the other issues encountered, especially with `gtest`?

boxanm, 2023-12-29:
We've just released [libpointmatcher 1.4.0](https://github.com/norlab-ulaval/libpointmatcher/releases/tag/1.4.0) which supports C++17!

closed 2023-12-29 by boxanm