norlab-ulaval/libpointmatcher #516: X-ICP code to detect and mitigate the degeneracy

Shelfcol, 2023-07-10:
Hello, thanks for your remarkable work.
I learned this repo has integrated the X-ICP module in your scan-to-map registration from paper X-ICP: Localizability-Aware LiDAR Registration for Robust Localization in Extreme Environments. But I cannot find this module. Could you please point the code's place? Or whether the module has been push to the repo?

boxanm, 2023-07-10:
Hello @Shelfcol and thanks for your interest in using libpointmatcher!
I believe that the work you mention has been done in one of (possibly private) libpointmatcher's forks, if not being directly part of ANYbotic's SLAM module. Maybe @YoshuaNava, one of the paper's coauthors, can confirm it.

Shelfcol, 2023-07-12:
@YoshuaNava 
Hello, I found a confusing part in your X-ICP paper, formulation 11 and 12: A's dimension is 3x3, while b's dimension is 1x1, so the optimization object seems not to be correct. Could you please point out where I understand wrongly? And has the optimization code been published open?
[image]

boxanm, 2023-07-12:
@Shelfcol Since this is no longer related to libpointmatcher, I suggest you contact the paper's authors directly. Consequently I'm closing this issue.

closed 2023-07-12 by boxanm