norlab-ulaval/libpointmatcher #526: Building error new release

dagata-mining, 2023-10-16:
Your last merge seems to be having issues on 20.04.06 Ubuntu at building. We used previous commit and it was working

/usr/bin/ld: /usr/local/lib/libpointmatcher.a(MatchersImpl.cpp.o): in function `MatchersImpl<double>::KDTreeMatcher::init(PointMatcher<double>::DataPoints const&)':
MatchersImpl.cpp:(.text._ZN12MatchersImplIdE13KDTreeMatcher4initERKN12PointMatcherIdE10DataPointsE[_ZN12MatchersImplIdE13KDTreeMatcher4initERKN12PointMatcherIdE10DataPointsE]+0x66): undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(MatchersImpl.cpp.o): in function `MatchersImpl<float>::KDTreeVarDistMatcher::init(PointMatcher<float>::DataPoints const&)':
MatchersImpl.cpp:(.text._ZN12MatchersImplIfE20KDTreeVarDistMatcher4initERKN12PointMatcherIfE10DataPointsE[_ZN12MatchersImplIfE20KDTreeVarDistMatcher4initERKN12PointMatcherIfE10DataPointsE]+0x66): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(MatchersImpl.cpp.o): in function `MatchersImpl<double>::KDTreeVarDistMatcher::init(PointMatcher<double>::DataPoints const&)':
MatchersImpl.cpp:(.text._ZN12MatchersImplIdE20KDTreeVarDistMatcher4initERKN12PointMatcherIdE10DataPointsE[_ZN12MatchersImplIdE20KDTreeVarDistMatcher4initERKN12PointMatcherIdE10DataPointsE]+0x66): undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(MatchersImpl.cpp.o): in function `MatchersImpl<float>::KDTreeMatcher::init(PointMatcher<float>::DataPoints const&)':
MatchersImpl.cpp:(.text._ZN12MatchersImplIfE13KDTreeMatcher4initERKN12PointMatcherIfE10DataPointsE[_ZN12MatchersImplIfE13KDTreeMatcher4initERKN12PointMatcherIfE10DataPointsE]+0x66): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(Saliency.cpp.o): in function `TensorVoting<float>::computeKnn(PointMatcher<float>::DataPoints const&)':
Saliency.cpp:(.text._ZN12TensorVotingIfE10computeKnnERKN12PointMatcherIfE10DataPointsE[_ZN12TensorVotingIfE10computeKnnERKN12PointMatcherIfE10DataPointsE]+0x8f): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
/usr/bin/ld: /usr/local/lib/libpointmatcher.a(Saliency.cpp.o): in function `TensorVoting<double>::computeKnn(PointMatcher<double>::DataPoints const&)':
Saliency.cpp:(.text._ZN12TensorVotingIdE10computeKnnERKN12PointMatcherIdE10DataPointsE[_ZN12TensorVotingIdE10computeKnnERKN12PointMatcherIdE10DataPointsE]+0x8f): undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
/usr/bin/ld: CMakeFiles/runQuickTest.dir/src/MapBuilding/Odometry/IcpOdometry.cpp.o: in function `KDTreeMatcherIntensity<float>::init(PointMatcher<float>::DataPoints const&)':
IcpOdometry.cpp:(.text._ZN22KDTreeMatcherIntensityIfE4initERKN12PointMatcherIfE10DataPointsE[_ZN22KDTreeMatcherIntensityIfE4initERKN12PointMatcherIfE10DataPointsE]+0x152): undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)

pomerlef, 2023-10-16:
@RedLeader962 it looks like libnabo is not linked. Is it on our side or it dates from a few PRs?

RedLeader962, 2023-10-16:
@pomerlef Seems related to PR #513  and issue #514. The build system is green on Ubuntu20.04 but I suppose the `libpointmatcher` install script used by the build system circumvent the issue by explicitly setting `LIBNABO_INSTALL_DIR` and `CMAKE_INSTALL_PREFIX`, i.e.:

```bash
cmake -D CMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE} \
  -D BUILD_TESTS=${BUILD_TESTS_FLAG} \
  -D GENERATE_API_DOC=${GENERATE_API_DOC_FLAG} \
  -D LIBNABO_INSTALL_DIR="${LPM_INSTALLED_LIBRARIES_PATH}/libnabo" \
  -D CMAKE_INSTALL_PREFIX="${LPM_INSTALLED_LIBRARIES_PATH}" \
  "${LPM_INSTALLED_LIBRARIES_PATH}/${LPM_LIBPOINTMATCHER_SRC_REPO_NAME}"
```

I'm pretty sure I encountered the same issue while implementing the build system hence this solution.

@dagata-mining Can you point out which commit work for you?

dagata-mining, 2023-10-17:
Yes might be an install problem, the libpointmatcher is building, but when used in other programs its crashing. 
We currently use 3464af6 in libpointmatcher and  c925c47 in libnabo . We did not versionned them in our build scripts so we just got the problem today.

Derrick1994, 2023-11-08:
I have encountered the same problem. May I ask if this issue has been resolved and if modifying the libnabo version can solve it?

RedLeader962, 2023-11-10:
@Derrick1994 Its not solved yet. We are on the verge of proceeding to the repository ownership transfer of [libnabo](https://github.com/ethz-asl/libnabo) very soon, integrate it to our build system and fix for good this issue. In the mean time, you can either use the [solution in my previous comment](https://github.com/norlab-ulaval/libpointmatcher/issues/526#issuecomment-1765391288) or use [both commit mentioned](https://github.com/norlab-ulaval/libpointmatcher/issues/526#issuecomment-1765574605) by @dagata-mining.

pomerlef, 2023-11-10:
As of today, libnabo is now hosted on Norlab's repo. We are progressing on this.

RedLeader962, 2023-12-06:
@dagata-mining @Derrick1994 We have pushed a fix for the install issue (Ref PR #538 and PR #539). The fix is currently only available on the `develop` branch at the moment and we are preparing a new release which will be available in the next few day.  
Thanks for flagging the issue and thanks your patience.

closed 2023-12-06 by RedLeader962