norlab-ulaval/libpointmatcher #537: std::shared_ptr<PointMatcher<float>::Matcher> kdtree

drhalftone, 2023-11-24:
So I'm trying to configure an ICP without YAML.  I have code working perfectly fine on Mac, but I have issues on Windows.  I know you don't support Windows, but this might be an easy fix for someone expert in using BOOST.  Here is my code:

    name = "KDTreeMatcher";
    parameters["knn"] = "1";
    parameters["epsilon"] = "3.16";
    std::shared_ptr<PointMatcher<float>::Matcher> kdtree = PointMatcher<float>::get().MatcherRegistrar.create(name, parameters);
    parameters.clear();
    
The problem is that forth line.  My compiler is throwing the error: 

lau3dsymmetryfilter.cpp:90:51: No viable conversion from 'PointMatcher<float>::Matcher *' to 'std::shared_ptr<PointMatcher<float>::Matcher>'
memory:1503:15: candidate constructor not viable: no known conversion from 'PointMatcher<float>::Matcher *' to 'nullptr_t' (aka 'std::nullptr_t') for 1st argument
memory:1559:5: candidate constructor not viable: no known conversion from 'PointMatcher<float>::Matcher *' to 'const shared_ptr<Matcher> &' for 1st argument
memory:1569:5: candidate constructor not viable: no known conversion from 'PointMatcher<float>::Matcher *' to 'shared_ptr<Matcher> &&' for 1st argument
memory:1564:5: candidate template ignored: could not match 'shared_ptr<_Ty2>' against 'PointMatcher<float>::Matcher *'
memory:1574:5: candidate template ignored: could not match 'shared_ptr<_Ty2>' against 'PointMatcher<float>::Matcher *'
memory:1587:5: candidate template ignored: could not match 'auto_ptr<_Ty2>' against 'PointMatcher<float>::Matcher *'
memory:1598:5: candidate template ignored: could not match 'unique_ptr<_Ux, _Dx>' against 'PointMatcher<float>::Matcher *'
memory:1509:14: explicit constructor is not a candidate
memory:1579:14: explicit constructor is not a candidate

I can fix this error if I use the following line of code:

    std::shared_ptr<PointMatcher<float>::Matcher> kdtree(PointMatcher<float>::get().MatcherRegistrar.create(name, parameters));

Regardless, I get an error when I assigned my ICP object's matcher field with the following line of code:

    icp.matcher = kdtree;
 
with the error:

lau3dsymmetryfilter.cpp:93:17: No viable overloaded '='
shared_ptr.hpp:584:18: candidate function not viable: no known conversion from 'std::shared_ptr<PointMatcher<float>::Matcher>' to 'const shared_ptr<Matcher>' for 1st argument
shared_ptr.hpp:593:18: candidate template ignored: could not match 'boost::shared_ptr' against 'std::shared_ptr'
shared_ptr.hpp:604:18: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
shared_ptr.hpp:613:18: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
shared_ptr.hpp:635:18: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
shared_ptr.hpp:644:18: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
shared_ptr.hpp:692:18: candidate function not viable: no known conversion from 'std::shared_ptr<PointMatcher<float>::Matcher>' to 'shared_ptr<Matcher>' for 1st argument
shared_ptr.hpp:699:18: candidate template ignored: could not match 'boost::shared_ptr' against 'std::shared_ptr'
shared_ptr.hpp:717:18: candidate function not viable: no known conversion from 'std::shared_ptr<PointMatcher<float>::Matcher>' to 'boost::detail::sp_nullptr_t' (aka 'std::nullptr_t') for 1st argument

drhalftone, 2023-11-24:
I replaced std::shared_ptr with boost::shared_ptr and that has taken the reported errors away.  I am now having other library issues that I hope are unrelated.  I will post my progress here.

drhalftone, 2023-11-24:
I think the above issue is fixed; however, I am having a lot of missing symbol errors like this:

lau3dsymmetryfilter.obj : error LNK2019: unresolved external symbol "public: __cdecl PointMatcher<float>::ErrorMinimizer::ErrorElements::ErrorElements(struct PointMatcher<float>::DataPoints const &,struct PointMatcher<float>::DataPoints,class Eigen::Matrix<float,-1,-1,0,-1,-1>,struct PointMatcher<float>::Matches)" (??0ErrorElements@ErrorMinimizer@?$PointMatcher@M@@QEAA@AEBUDataPoints@2@U32@V?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@UMatches@2@@Z) referenced in function "public: __cdecl LAU3DSymmetryFilter::LAU3DSymmetryFilter(int,int,enum LAU3DSymmetryFilter::Axis,class QObject *)" (??0LAU3DSymmetryFilter@@QEAA@HHW4Axis@0@PEAVQObject@@@Z)
pointmatcher.lib(MatchersImpl.obj) : error LNK2019: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<float,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@MV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z) referenced in function "public: virtual void __cdecl MatchersImpl<float>::KDTreeMatcher::init(struct PointMatcher<float>::DataPoints const &)" (?init@KDTreeMatcher@?$MatchersImpl@M@@UEAAXAEBUDataPoints@?$PointMatcher@M@@@Z)
pointmatcher.lib(SpectralDecomposition.obj) : error LNK2001: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<float,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@MV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z)
pointmatcher.lib(Saliency.obj) : error LNK2001: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<float,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@MV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z)
pointmatcher.lib(MatchersImpl.obj) : error LNK2019: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<double,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@NV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z) referenced in function "public: virtual void __cdecl MatchersImpl<double>::KDTreeMatcher::init(struct PointMatcher<double>::DataPoints const &)" (?init@KDTreeMatcher@?$MatchersImpl@N@@UEAAXAEBUDataPoints@?$PointMatcher@N@@@Z)
pointmatcher.lib(SpectralDecomposition.obj) : error LNK2001: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<double,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@NV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z)
pointmatcher.lib(Saliency.obj) : error LNK2001: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<double,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<double,class Eigen::Matrix<double,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@NV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z)
release\LAUSymmetryFilter.exe : fatal error LNK1120: 4 unresolved externals
jom: C:\Users\dllau\Developer\LAUCowCommandLineTools\build-LAUSymmetryFilter-Desktop_x86_windows_msvc2019_pe_64bit-Release\Makefile.Release [release\LAUSymmetryFilter.exe] Error 1120
jom: C:\Users\dllau\Developer\LAUCowCommandLineTools\build-LAUSymmetryFilter-Desktop_x86_windows_msvc2019_pe_64bit-Release\Makefile [release] Error 2
13:11:27: The process "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" exited with code 2.
Error while building/deploying project LAUSymmetryFilter (kit: Desktop (x86-windows-msvc2019-pe-64bit))
When executing step "Make"
13:11:27: Elapsed time: 00:05.

boxanm, 2023-11-25:
Hey @drhalftone,
As you said, we don't support Windows at the moment. However, in case you figure out how to fix the missing symbols errors, don't hesitate to submit a PR! Our Windows build test infrastructure is not in place yet, but we will process it eventually.