norlab-ulaval/libpointmatcher #551: Bug: CMake error on doxygen doc compilation

RedLeader962, 2024-01-10:
CMake compilation failure when trying to compile libpointmatcher doxygen doc via libpointmatcher-build-system

### Step to reproduce

- OS: Ubuntu 
- version: bionic, focal
- Arch: arm64
- Software version: libpointmatcher-build-system latest 

#### Command
```shell
cmake -D CMAKE_BUILD_TYPE=Release -D BUILD_TESTS=TRUE -D GENERATE_API_DOC=TRUE /opt/percep3d_libraries/libpointmatcher
```

#### Result
```shell
0.166 -- The C compiler identification is GNU 9.4.0
0.208 -- The CXX compiler identification is GNU 9.4.0
0.211 -- Check for working C compiler: /usr/bin/cc
0.256 -- Check for working C compiler: /usr/bin/cc -- works
0.257 -- Detecting C compiler ABI info
0.299 -- Detecting C compiler ABI info - done
0.306 -- Detecting C compile features
0.307 -- Detecting C compile features - done
0.308 -- Check for working CXX compiler: /usr/bin/c++
0.361 -- Check for working CXX compiler: /usr/bin/c++ -- works
0.362 -- Detecting CXX compiler ABI info
0.413 -- Detecting CXX compiler ABI info - done
0.421 -- Detecting CXX compile features
0.421 -- Detecting CXX compile features - done
0.423 -- Looking for pthread.h
0.472 -- Looking for pthread.h - found
0.473 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
0.517 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
0.517 -- Looking for pthread_create in pthreads
0.555 -- Looking for pthread_create in pthreads - not found
0.555 -- Looking for pthread_create in pthread
0.604 -- Looking for pthread_create in pthread - found
0.604 -- Found Threads: TRUE  
0.615 -- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: thread filesystem system program_options date_time 
0.621 -- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: thread filesystem system program_options date_time chrono 
0.623 -- libnabo found, version 1.0.7 (Config mode,  libs=libnabo::nabo)
0.623 -- Looking for yaml-cpp on system...
0.624 -- Looking for _POSIX_TIMERS
0.669 -- Looking for _POSIX_TIMERS - found
0.671 -- API Documentation (doxygen): enabled
0.697 -- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
0.720 -- Found DOXYFILE_IN: /opt/percep3d_libraries/libpointmatcher/Doxyfile.in  
0.722 -- Found LATEX: /usr/bin/latex   
0.723 CMake Error at UseDoxygen.cmake:112 (get_target_property):
0.723   get_target_property() called with non-existent target "doc".
0.723 Call Stack (most recent call first):
0.723   CMakeLists.txt:433 (include)
0.723 
0.723 
0.728 -- Configuring incomplete, errors occurred!
0.728 See also "/opt/percep3d_libraries/libpointmatcher/build/CMakeFiles/CMakeOutput.log".
0.728 See also "/opt/percep3d_libraries/libpointmatcher/build/CMakeFiles/CMakeError.log".

```

### Investigation

Is it possible that  the following code in `UseDoxygen.cmake` at line 112
```
get_target_property(DOC_TARGET doc TYPE)
if(NOT DOC_TARGET)
    add_custom_target(doc)
endif()
```

should be 
```
if(NOT TARGET doc)
    add_custom_target(doc)
endif()
```

boxanm, 2024-01-22:
Nice catch! Fails also on MacOS Sonoma.

RedLeader962, 2024-01-22:
In progress in branch [dev--fix-Bug-CMake-error-on-doxygen-doc-compilation-(github-issue-551)-NMO-478](https://github.com/norlab-ulaval/libpointmatcher/tree/dev--fix-Bug-CMake-error-on-doxygen-doc-compilation-(github-issue-551)-NMO-478)

boxanm, 2024-05-23:
Fixed in https://github.com/norlab-ulaval/libpointmatcher/pull/568

closed 2024-05-23 by boxanm