norlab-ulaval/libpointmatcher #563: find_dependency(yaml-cpp REQUIRED)

lrineau, 2024-04-30:
`yaml-cpp` is a required dependency:

https://github.com/norlab-ulaval/libpointmatcher/blob/cb16b20759d80f3ebe33736da5d70fd8491edb58/CMakeLists.txt#L220-L221

It should be added as a dependency in `libpointmatcherConfig.cmake.in`, like `libnano`:

https://github.com/norlab-ulaval/libpointmatcher/blob/cb16b20759d80f3ebe33736da5d70fd8491edb58/libpointmatcherConfig.cmake.in#L7-L12

My proposed patch is add

```cmake
 find_dependency(yaml-cpp REQUIRED)
```

after line 8 of `libpointmatcherConfig.cmake.in`.

Without that, a CMake user of `libpointmatcherConfig.cmake` has a CMake error:

```
CMake Error at Installation/cmake/modules/CGAL_pointmatcher_support.cmake:12 (target_link_libraries):
  The link interface of target "CGAL::pointmatcher_support" contains:

    yaml-cpp::yaml-cpp

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
```

see a faulty CI testsuite log at https://github.com/CGAL/cgal-testsuite-dockerfiles/actions/runs/8891254550/job/24412843289#step:3:10198.

For information, the target `CGAL::pointmatcher_support` is created  by the file `CGAL_pointmatcher_support.cmake` of CGAL, that contains:

```cmake
if(libpointmatcher_FOUND AND NOT TARGET CGAL::pointmatcher_support)
  find_package(Boost COMPONENTS thread filesystem system program_options date_time chrono)
  if(Boost_chrono_FOUND
      AND Boost_thread_FOUND
      AND Boost_filesystem_FOUND
      AND Boost_system_FOUND
      AND Boost_program_options_FOUND
      AND Boost_date_time_FOUND)
    add_library(CGAL::pointmatcher_support INTERFACE IMPORTED)
    target_compile_definitions(CGAL::pointmatcher_support INTERFACE "CGAL_LINKED_WITH_POINTMATCHER")
    target_include_directories(CGAL::pointmatcher_support INTERFACE "${libpointmatcher_INCLUDE_DIR}")
    target_link_libraries(CGAL::pointmatcher_support INTERFACE ${libpointmatcher_LIBRARIES} libnabo::nabo)
  else()
    message(STATUS "NOTICE: the libpointmatcher library requires the following boost components: thread filesystem system program_options date_time chrono.")
  endif()
endif()
```

cc: @soesau  We have a CMake error with the new released Ubuntu 24.084, that ships `yaml-cpp` version 0.8.0.

ryancu7, 2024-11-27:
Either that, or yaml-cpp goes back to not being REQUIRED, which I would prefer. 4f8de80450cbbd480dc85d950a0d1087970cc1bf made it required.

Note that pointmatcher/Documentation.dox still currently speaks of it being an option, even though it currently isn't:
> **If** you have compiled libpointmatcher with yaml-cpp...

boxanm, 2024-12-06:
Hi @ryancu7!
thanks for pointing out the outdated documentation. May I ask what is your use case for `libpointmatcher`? We typically see users configuring lpm with yaml files.

ryancu7, 2024-12-09:
We are hoping to develop our own Windows-based GUI for any registration options as part of our commercial application and the end users will not be interfacing with YAML. If our code would need to generate YAML to send to libpointmatcher, this would be trickier and less typesafe than us using C++ directly, and for us, not worth the hassle of bundling the extra dependency. 

doc/ApplyingDataFilters.md also speaks of YAML being optional. 
Thanks for making your work available.

closed 2024-12-09 by boxanm