norlab-ulaval/libpointmatcher #571: Compilation error with MSVC compiler

chetan-sadhu, 2024-05-17:
When trying to compile with Microsoft Visual Studio 2017 compiler with `USE_OPEN_MP = ON`, we get `Compiler Error C3016` because OpenMP for loops uses std::size_t as a the datatype for loop index.

boxanm, 2024-05-18:
Hi @chetan-sadhu 
we currently don't have the resources needed to support Windows OS. However, if you are confident with your workaround, don't hesitate to submit a PR and update the [Windows guide](https://github.com/norlab-ulaval/libpointmatcher/blob/master/doc/CompilationWindows.md) accordingly!
Alternatively, you can also build and run `libpointmatcher` in [WSL](https://learn.microsoft.com/en-us/windows/wsl/about) which should work seamlessly.

lingerer, 2024-12-02:
You can set /openmp:llvm under vs 2022