norlab-ulaval/libpointmatcher #585: PointMatcherSupport::timer seems off

Louis-ThomasSchreiber, 2024-08-29:
I tried using the default PerformanceInspector and the FileLogger available in libpointmatcher. The output times looked off: The mean iteration time x number of iterations is 10x the actual execution time. I tried investigating with this simple test:
```
const Time::V time1 = Time::Clock::now();
PointMatcherSupport::timer t1;
t1.restart();

this_thread::sleep_for(chrono::seconds(10));

const Time::V time2 = Time::Clock::now();

std::cout << "PM:timer time = " << t1.elapsed() << " chrono duration = "
          << std::chrono::duration_cast<std::chrono::milliseconds>(time2 - time1).count() << std::endl;
```
which outputs:
PM:timer time = 4.9282e-05 chrono duration = 10000

Is there something I dont understand?

boxanm, 2024-08-31:
Hello @Louis-ThomasSchreiber !
I tested your example, and everything seems functional on my side, as I'm getting the correct results:
```text
╰─$ ./examples/timer
PM:timer time = 10.0046 chrono duration = 10004 ms
```
Can you provide more information about your system? If you are on Ubuntu or MacOS, you run one of the following scripts: `utest/listVersionsUbuntu.sh` or `utest/listVersionsMacOS.sh`.

Louis-ThomasSchreiber, 2024-11-07:
Im running Unutu 20.04 in a docker container, my PC runs Ubuntu 22.04.

Name            | Version
----------------|-------------------------------
ubuntu:           | Description: Ubuntu 20.04.6 LTS
architecture:   | 64-bit
gcc:            | gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
git:            | git version 2.25.1
cmake:          | cmake version 3.16.3 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost:          | Version: 1.71.0.0ubuntu2
eigen3:         | Version: 3.3.7-2
dpkg-query: package 'doxygen' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
doxygen:        |

boxanm, 2025-01-12:
@Louis-ThomasSchreiber can you please confirm that you see the issue when using non-POSIX timers? See this recent PR: https://github.com/norlab-ulaval/libpointmatcher/pull/598

Louis-ThomasSchreiber, 2025-02-13:
I cant confirm since I was checked out on a version of libpointmatcher from 2022.
Very probable that the issue is fixed, thanks.

closed 2025-02-13 by Louis-ThomasSchreiber