norlab-ulaval/libpointmatcher #603: libpointmatcher RobustOutlierFilter L1

justsoso-ytt, 2025-05-06:
./icp \
  --config "L1.yaml" \
  --output "result-t" \
  -v \
  "../MountainPlain/Hokuyo_0.csv" \
  "../MountainPlain/Hokuyo_1.csv"
match ratio: 30.7748
ICP transformation:
   0.934511  -0.0199831    0.355371  -0.0126413
  0.0320857    0.999088  -0.0281947  0.00694489
  -0.354484   0.0377506      0.9343 -0.00781989
          0           0           0           1
为什么会match ratio: 30.7748>1

justsoso-ytt, 2025-05-06:
---
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.5

referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 10

matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 0

outlierFilters:
  - MaxDistOutlierFilter:
      maxDist: 0.5  
  - RobustOutlierFilter:
      robustFct: "L1"              # 使用 L1 范数（无参数，论文结论显示其稳定性）
      tuning: 0.1                   # 占位参数（L1 无需 tuning，但需保留字段）
      scaleEstimator: "mad"        # MAD 尺度估计（Median Absolute Deviation，论文推荐）
      nbIterationForScale: 0      # 允许 1 次尺度迭代（论文实验表明动态更新提升鲁棒性）
      distanceType: "point2point"  # 点对面误差（论文表 II 配置，优于点对点）

errorMinimizer:
  PointToPlaneErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 800000
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.0001
      minDiffTransErr: 0.001
      smoothLength: 4

# inspector:
#   NullInspector

inspector:
  VTKFileInspector:
    baseFileName: pointmatcher-run1
    dumpPerfOnExit: 0
    dumpStats: 0
    dumpIterationInfo: 1
    dumpDataLinks: 0
    dumpReading: 0
    dumpReference: 0

logger:
  NullLogger
#  FileLogger
This is my yaml setting

boxanm, 2025-05-06:
Hi @justsoso-ytt , can you describe what is the issue you experience?

justsoso-ytt, 2025-05-06:
> 你好[@justsoso-ytt](https://github.com/justsoso-ytt)，您能描述一下您遇到的问题吗？
This is the result of the huber kernel function
./pmicp \
  --config "Huber.yaml" \
  --output "result-t" \
  -v \
  "../Tree/Hokuyo_0.csv" \
  "../Tree/Hokuyo_1.csv"
match ratio: 0.889034
ICP transformation:
   0.991746    -0.12716   0.0164122    0.156696
   0.127082    0.991875  0.00573883  -0.0289652
 -0.0170086 -0.00360584    0.999849   0.0101571
          0           0           0           1
This is the result of the L1 kernel function
./pmicp \
  --config "L1.yaml" \
  --output "result-t" \
  -v \
  "../Tree/Hokuyo_0.csv" \
  "../Tree/Hokuyo_1.csv"
match ratio: 117.591
ICP transformation:
    0.999458   -0.0328551   0.00200086   0.00652048
   0.0328592     0.999458  -0.00207124 -0.000140734
 -0.00193179   0.00213582     0.999996  -0.00199306
           0            0            0            1

When I run icp and OutlierFiltersFamilies fiLter in the yamL fiLe as RobustOutlierFilter and the kerneL function is L 1, match ratio outputs a vaLue of > 1, but the rest of the kerneL functions do not